libimobiledevice/libimobiledevice #307: Not device connection

elfrock, 2016-04-17:
Hello!
I have a problem with my iphone connection. Till march it was ok but suddenly something went wrong.
i use archlinux, kernel 4.4.5-1 and latest packages:
libimobiledevice-1.2.0
libplist-1.12
libusbmuxd-1.0.10
usbmuxd-1.1.0

Then I try:
`usbmuxd -f -v`
I got:
> Could not open lockfile

If I do it like `sudo usbmuxd -f -v`
I got:

> [08:46:01.296][3] usbmuxd v1.1.1 starting up
> [08:46:01.296][4] Creating socket
> [08:46:01.297][4] Initializing USB
> [08:46:01.302][4] Registering for libusb hotplug events
> [08:46:01.302][4] Found new device with v/p 05ac:1290 at 3-12
> [08:46:01.302][4] Found interface 1 with endpoints 04/85 for device 3-12
> [08:46:01.303][4] Using wMaxPacketSize=512 for device 3-12
> [08:46:01.303][4] USB Speed is 480 MBit/s for device 3-12
> [08:46:01.303][3] Connecting to new device on location 0x3000c as ID 1
> [08:46:01.304][4] 1 device detected
> [08:46:01.304][3] Initialization complete
> [08:46:01.304][4] New client on fd 12 (pid 29528)
> [08:46:01.304][3] Connected to v2.0 device 1 on location 0x3000c with serial number 0d6bdd61342106d5e21144d4e2a64c547ccfaa21
> [08:46:01.304][4] preflight_worker_handle_device_add: Starting preflight on device 0d6bdd61342106d5e21144d4e2a64c547ccfaa21...
> [08:46:01.305][4] New client on fd 14 (self)
> [08:46:01.308][4] device_control_input: AppleUSBDeviceMux::handleConnectResult new session 0xc2b3f780 established 62078<-lo0->49224 62078<-usb->1
> [08:46:01.314][4] New client on fd 16 (self)
> [08:46:01.314][4] Client 16 connection closed
> [08:46:01.314][4] Disconnecting client fd 16
> [08:46:01.321][4] New client on fd 16 (self)
> [08:46:01.321][4] Client 16 connection closed
> [08:46:01.321][4] Disconnecting client fd 16
> [08:46:02.001][4] New client on fd 15 (pid 1525)
> [08:46:02.102][4] Client 15 connection closed
> [08:46:02.102][4] Disconnecting client fd 15
> [08:46:03.001][4] New client on fd 15 (pid 1525)
> [08:46:03.101][4] Client 15 connection closed
> [08:46:03.102][4] Disconnecting client fd 15
> [08:46:04.001][4] New client on fd 15 (pid 1525)
> [08:46:04.102][4] Client 15 connection closed
> [08:46:04.102][4] Disconnecting client fd 15
> [08:46:05.000][4] New client on fd 15 (pid 1525)
> [08:46:05.101][4] Client 15 connection closed
> [08:46:05.101][4] Disconnecting client fd 15

It looks like cycle of connection and disconnection.
Could you help me please with this issue?

Lipovlan, 2018-06-04:
Have you tried inputting your password on your iPhone?

mexmer, 2018-06-05:
looks like cable or port issue, device is reconnecting even before handshake can be persormed, only once you can see preflight attempt, but that did't pass.

restart phone, that can also help, if it's not cable issue.

Fr33maan, 2019-11-29:
I come from here [remotedebug-ios](https://github.com/RemoteDebug/remotedebug-ios-webkit-adapter/issues/171) and by trying all dependencies tools, the issues appeared to be usbmuxd deamon which was unable to start because of lockfile. I was able to start it with sudo and then the remote debuger works.

ojnickel, 2021-01-27:
could you pls elaBoraTe a bit more and tell, how exactly dis you remove      this lockfile?