libimobiledevice/libimobiledevice #311: [iOS 9.3] unback stuck on 'Device is not ready yet. Going to try again in 2 seconds...'

norman784, 2016-04-21:
I've an issue with iOS 9.3 but stuck on 'Device is not ready yet...', there are 3 options that could be:

- iOS closes the connection after reading 5 applications
- iOS don't let them read TestFlight data
- iOS don't let them read the next app that comes after TestFlight

The partial log looks like:

```
15:30:58 property_list_service.c:191 internal_plist_receive_timeout(): initial read=4
15:30:58 property_list_service.c:200 internal_plist_receive_timeout(): 206 bytes following
15:30:58 property_list_service.c:213 internal_plist_receive_timeout(): received 206 bytes
15:30:58 property_list_service.c:239 internal_plist_receive_timeout(): printing 408 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
  <string>DLMessageCreateDirectory</string>
  <string>_unback_/__UUID__/var/mobile/Applications/com.armorgames.kingdomrushiphone/Documents</string>
  <dict>
  </dict>
  <real>0.000000</real>
</array>
</plist>
15:30:58 property_list_service.c:127 internal_plist_send(): sending 106 bytes
15:30:58 service.c:140 service_send(): sending 4 bytes
15:30:58 service.c:140 service_send(): sending 106 bytes
15:30:58 property_list_service.c:132 internal_plist_send(): sent 106 bytes
15:30:58 property_list_service.c:133 internal_plist_send(): printing 344 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
  <string>DLMessageStatusResponse</string>
  <integer>0</integer>
  <string>___EmptyParameterString___</string>
  <string>___EmptyParameterString___</string>
</array>
</plist>
15:30:58 property_list_service.c:191 internal_plist_receive_timeout(): initial read=4
15:30:58 property_list_service.c:200 internal_plist_receive_timeout(): 184 bytes following
15:30:58 property_list_service.c:213 internal_plist_receive_timeout(): received 184 bytes
15:30:58 property_list_service.c:239 internal_plist_receive_timeout(): printing 386 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
  <string>DLMessageCreateDirectory</string>
  <string>_unback_/__UUID__/var/mobile/Applications/com.apple.TestFlight</string>
  <dict>
  </dict>
  <real>0.000000</real>
</array>
</plist>
15:30:58 property_list_service.c:127 internal_plist_send(): sending 106 bytes
15:30:58 service.c:140 service_send(): sending 4 bytes
15:30:58 service.c:140 service_send(): sending 106 bytes
15:30:58 property_list_service.c:132 internal_plist_send(): sent 106 bytes
15:30:58 property_list_service.c:133 internal_plist_send(): printing 344 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
  <string>DLMessageStatusResponse</string>
  <integer>0</integer>
  <string>___EmptyParameterString___</string>
  <string>___EmptyParameterString___</string>
</array>
</plist>
15:30:58 property_list_service.c:191 internal_plist_receive_timeout(): initial read=4
[... 169 lines cut ...]
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
  <string>DLMessageStatusResponse</string>
  <integer>0</integer>
  <string>___EmptyParameterString___</string>
  <string>___EmptyParameterString___</string>
</array>
</plist>
15:30:58 idevice.c:379 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -35 (Resource temporarily unavailable)
15:30:58 service.c:163 service_receive_with_timeout(): could not read data
15:30:58 property_list_service.c:191 internal_plist_receive_timeout(): initial read=0
15:30:58 property_list_service.c:193 internal_plist_receive_timeout(): initial read failed!
Device is not ready yet. Going to try again in 2 seconds...
15:30:59 notification_proxy.c:48 np_lock(): Locked
15:30:59 service.c:163 service_receive_with_timeout(): could not read data
15:30:59 notification_proxy.c:260 np_get_notification(): NotificationProxy: no notification received!
15:30:59 notification_proxy.c:59 np_unlock(): Unlocked
15:31:00 idevice.c:379 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -35 (Resource temporarily unavailable)
15:31:00 service.c:163 service_receive_with_timeout(): could not read data
15:31:00 property_list_service.c:191 internal_plist_receive_timeout(): initial read=0
15:31:00 property_list_service.c:193 internal_plist_receive_timeout(): initial read failed!
Device is not ready yet. Going to try again in 2 seconds...
```

morgrowe, 2016-07-05:
I'm having the same issue. Did you ever resolve it, @norman784?

rcmpayne, 2016-07-05:
unback feature is broken/not supported on newer device's.

norman784, 2016-07-05:
I'm using a ruby script to unback the backup (this is a similar [python script](http://stackoverflow.com/questions/1498342/how-to-decrypt-an-encrypted-apple-itunes-iphone-backup))

nikias, 2016-07-05:
Apple either accidentally or intentionally broke the unback command. This is not a bug in libimobiledevice.

norman784, 2016-07-05:
good to know that, thanks @nikias

morgrowe, 2016-07-05:
Thanks for the info. Shame Apple decided to bork this functionality.

@norman784 I had a quick look for the tool mentioned in the answer of the page you linked, but it appears it's [no longer available](https://code.google.com/archive/p/iphone-dataprotection/downloads). Is the ruby script you use something you're able to share?

norman784, 2016-07-06:
@morgrowe it took me a while to find the original post from where I brought, heres the [python](http://stackoverflow.com/a/3130860/825137) and [ruby](http://stackoverflow.com/a/9253197/825137) implementation

inflex, 2016-07-23:
Glad to see I'm not going insane, this has driven me up the wall the last few weeks and I even installed/built-from-git-source on 3 different machine to no avail :-1: 

Will there be a feature add implementing a replacement unback facility built in to the idevicebackup2 tool to achieve the same outcome?

rcmpayne, 2016-07-23:
Has anyone got the script to work on Windows? I can get it going.

inflex, 2016-10-03:
Until idevicebackup2 or apple implements the functionality again, I've written a decoder in C for linux (probably compiles on others).  It doesn't handle encrypted backups though.

https://github.com/inflex/ideviceunback