libimobiledevice/libimobiledevice #313: How to push any file to the ios device

vertissachin, 2016-05-03:
Hi,

 I have to push a log file to the device like we are able to push file in android using adb , is there any way to push log file in a device . actulay i wants to push a empt file in device and have to capture the logs of application in that log file.

shivam3112, 2016-05-07:
Hi @vertissachin  there is no backdoor or bridge for pushing any file in iOS devices.

mexmer, 2016-05-08:
if it's your application we talking about, then you can utilize document sharing for that. 

of course if you want to do this for non owned or system app, it will not be possible without jailbreak and afc2 service enabled.

eadmaster, 2021-03-27:
I'm also interested in pulling files in addition to pulling.

Maybe some code can be reused from the [ifuse](https://github.com/libimobiledevice/ifuse) project?

mexmer, 2021-03-29:
as i mentioned before, you can only push/pull files from document sharing, you can also pull photos from device.
and yes ifuse supports that.

only if you have jailbroken phone with AFC2 service enabled, you can read and write any file on phone, that is not in use or protected.

eadmaster, 2021-03-29:
ok, i don't how ifuse does that, but can we use the same method here to push+pull files that are accessible on non-jailbroken phones?

It think this would be handy if you only need to trasfer a few files and do not want to install another tool and/or mount the volume.

mexmer, 2021-04-01:
like i said, ifuse supports that. you can push/pull files even on nonjailbroken phone, but on nonjailbroken phone you are limited to camera photos, icloud photo cache and itunes media files (when it comes to pull), you can try to push files into ios device media filesystem, but it will not be recognized by phone, because you need to do proper synchronization (eg. emulate itunes).

how to use ifuse is described on ifuse page
https://github.com/libimobiledevice/ifuse

eadmaster, 2021-04-16:
> you need to understand one simple thing, it's not libimobiledevice library, that limits your access, it's device operating system (eg. iOS)

I have no problem with restricted FS access, actually here i was suggesting adding the same feature in `libimobiledevice`

mexmer, 2021-04-16:
i see now, yes it might come handy sometimes, but don't forget you will still not have access to files you push into device, they will not be visible in ios file manager, it simply doesn't work that way.

for multiple files mounting iphone with ifuse is still better option tho'. or even if you plan to use your ios device as removable storage. (mind ifuse is not available on windows, so you will not be able to access those files on windows, except with special software, that can list content of AFC)