libimobiledevice/libimobiledevice #318: Relative paths (to make it portable)

norman784, 2016-05-18:
I'm curious if its posible to make this library portable, b/c now if I move the bin to another folder it stop working, the build script looks like:

```
LIB_PATH="libimobiledevice"

[ -d "$LIB_PATH" ] && rm -rf $LIB_PATH

mkdir $LIB_PATH

cd $LIB_PATH

LIB_PATH=${PWD}
PKG_CONFIG_PATH="$LIB_PATH/lib/pkgconfig"

echo "Paths:"
echo "-------------------------------------------"
echo "  [installation path] $LIB_PATH"
echo "  [pkgconfig path]    $PKG_CONFIG_PATH"
echo ""

cd ..
[ -d "tmp" ] && rm -rf "tmp"
mkdir tmp && cd tmp
cd tmp

echo "Building libplist..."
echo "-------------------------------------------"
echo ""
git clone https://github.com/libimobiledevice/libplist.git
cd libplist
PKG_CONFIG_PATH="$LIB_PATH/lib/pkgconfig" ./autogen.sh
PKG_CONFIG_PATH="$LIB_PATH/lib/pkgconfig" ./configure --prefix=$LIB_PATH #--enable-debug-code
make
make install

cd ..

echo ""
echo "Building usbmuxd..."
echo "-------------------------------------------"
echo ""
git clone https://github.com/libimobiledevice/libusbmuxd.git
cd libusbmuxd
PKG_CONFIG_PATH="$LIB_PATH/lib/pkgconfig" ./autogen.sh
PKG_CONFIG_PATH="$LIB_PATH/lib/pkgconfig" ./configure --prefix=$LIB_PATH #--enable-debug-code
make
make install

cd ..

echo ""
echo "Building libimobiledevice..."
echo "-------------------------------------------"
echo ""
git clone https://github.com/libimobiledevice/libimobiledevice.git
cd libimobiledevice
PKG_CONFIG_PATH="$LIB_PATH/lib/pkgconfig" ./autogen.sh
PKG_CONFIG_PATH="$LIB_PATH/lib/pkgconfig" ./configure --prefix=$LIB_PATH #--enable-debug-code
make
make install

echo ""
echo "...cleaning"
cd ..
rm -rf tmp
```

Whit this script I can build it but if I change *LIB_PATH* from `LIB_PATH=${PWD}` to something like this `LIB_PATH='.'` it stop working, what I need/want to accomplish its to ship the bin with an osx .app, I could just use the default `/Applications/MyApp.app/....` but the user can run the app from anywhere (just like an exe)

Regards

mexmer, 2016-05-19:
it's not issue of build, but how shared libraries works in linux.
set 
LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$directorywithlibimobilelibrary
prior starting any command that uses library

if location is "fixed" although nonstandard, you can set
LD_PRELOAD_PATH and run ldconfig instead, but that would probly be good for portable intention.

for further information regarding shared libraries on linux see
http://tldp.org/HOWTO/Program-Library-HOWTO/shared-libraries.html

norman784, 2016-05-19:
thanks, will take a look