libimobiledevice/libimobiledevice #322: how to backup contacts by libimobiledevice?

lianzisong, 2016-05-27:
Does libimobiledevice support it?
otherwise ,how to do it?

gitsop01, 2016-05-28:
Contacts Backup can be achieved thru  "ICoud " when active, you will have to log into your account on ICloud and download your contacts in vcard format.

Not sure if an open source application exists to backup directly with Linux or Microsoft but is possible
with this library. Private and commercial applications most likely exist.

A development  linux command line gtk3 updated program from evolution-data-server to idevice for one-way contacts sync is possible using this open-source code at:

https://github.com/gitsop01/eds-to-idevice

This code could be modified to achieve a direct back-up of contacts with a bit of programming effort.

lianzisong, 2016-05-28:
@gitsop01 ,Thanks,  I've read the "ds-to-idevice" project.  I found that the Contacts backup has two methods, 1st is , backup the IOS data, then parse the Contacts's db ,2nd is synchronize the Contacts ..am I right?
if so , I wonder if I set the backup password ,Can I still get the unencrypted Contacts data by synchronization?

gitsop01, 2016-05-28:
No, eds-to-idevice does not backup any ios data at all.

Unfortunately eds-to-idevice reads the evolution-data-server contact info, then creates multiple plist info in the correct format to be able to sync(write) it to an idevice. It is only one way from eds to an idevice. It can delete all idevice contact info and then sync the eds contact data to the idevice. 

eds-to-idevice output the first and last name to the terminal of each contact it finds in eds to check
what is being read, and to be able to manually check for duplicate info or contact info you may want to delete, but you will have to use either evolution or gnome-contacts to actually do the editing or delete of the contact info in evolution-data-server, then delete all idevice contact info and resync contact to the idevice.

What you need is to be able to read the idevice contact info using the correct plist info format for all the fields available in each contact info, and write them out as either a Vcard format or comma separated fields that can be used as a backup file. Vcard info can be used to transfer contact info to many contact managers and email client programs, and mobile phones using third part apps or directly.

eds-to idevice can output xml list for device communication to an idevice but it is not reliable in the testing I have done and needs work and trouble-shooting before it will be reliable. 

Two-way sync with eds-to-idevice will require a fair bit of programming and testing to get it to build the
plist to read the idevice contact fields and map them to a vcard format directly or write them to eds
and use evolution to create the vcard format file as a backup.  Basically do the reverse of what the 
program does at the moment, but two-way syncs actually have to do both operations and check for
duplicate entries etc and handle these situations correctly which can add a fair bit of complexity as you may have duplicate contacts that alter by one field or many and you have to decide how you are going to handle each situation.

Gnome-contacts can do two-way syncs to eds and google contacts but it is just creating the contacts db in eds and you can transfer contacts between the db's in eds. then eds backend will sync it back
to the actual contacts list. If you have duplicates contacts names or create duplicates contacts names
it will ask if you want to merge the info into one contact but I think it only does it using the name fields
only.

lianzisong, 2016-05-30:
@gitsop01 thanks. I'm confused by your comments now . 
1.U told me that "eds-to-idevice "  doesn't backup the contacts at all .it only synchronize the  local contacts info to the idevice. am I right?
2.But what I found from the "eds-to-idevice " ,there is a function named " eti_sync_get_contacts(..)" ,and the detail of the function really do the synchronization,  doesn't it get the contacts info from the idevice?  what exactly work doest this function do ? Can u show me the detail? Thanks.

gitsop01, 2016-05-30:
Yes- but is sync's default addressbook contacts from evolution-data-server to an idevice. One way only.

eti_sync_get_contacts is setting up the sync and parsing the contacts info obtained from evolution-data-server to the idevice. Not the other way around.

lianzisong, 2016-05-30:
@gitsop01  thanks , so the conclusion is :  the libimobiledevice doesn't support the the function getting the contacts from idevice?

gitsop01, 2016-05-30:
Libmobiledevice library has the support functions to be able to do it, but an application is not included to do it. 
As far as ltools No.

lianzisong, 2016-05-31:
@gitsop01 thanks.
at least I know the libimobiledevice could support my need.
BTW, Can you offer me the example of Contacts backup ,any project in the github is OK too.
I don't know how to build the logic of Contact backup.

Kanong017, 2016-05-31:
厦门的?给个联系方式呗一起交流下libimobiledevice

lianzisong, 2016-05-31:
@Kanong017 lianzss1987@gmail.com..

lianzisong, 2016-05-31:
@Kanong017 用另一个吧。我这里gmail又被墙了。
lianzss@163.com

gitsop01, 2016-05-31:
No, I do not know of any open-source projects but the the wiki for libmobiledevice list it as WIP
but no other info.

lianzisong, 2016-06-01:
@gitsop01 Thanks ,anyway.

closed 2016-09-07 by lianzisong