libimobiledevice/libimobiledevice #324: How to use tools to install .mobileconfig

rcmpayne, 2016-06-07:
is there a way to use these tools to install a mobileconfig profile? Currently, I am sending them via attachment to mail app and selecting them to load from that.

rcmpayne, 2016-06-08:
When i use 9.3.1 iOS device i get 0xe800800d 

	{ 0xe800800d, "There was an error communicating with your device." },

if i use a older device pre (9.2.1) to push a profile i get  0xe8008013

	{ 0xe8008013, "This provisioning profile does not have a valid signature (or it has a valid, but untrusted signature)." },

Kanong017, 2016-06-25:
i have a tool can install .mobileconfig,if you wanna get,Email:820957570@qq.com

DaKaZ, 2016-06-25:
@rcmpayne could you please share how you installed the mobileconfig file prior to 9.3.1?

rcmpayne, 2016-06-27:
@DaKaZ , you cant. pre 9.3.1 you can executethe -l "list command" and it works but you cant install on any version from my testing (iOS 8+)

DaKaZ, 2016-06-27:
Thanks @rcmpayne. I figure it has to be possible since you can install the mobile config from Apple Configurator. Any idea if there is a roadmap / todo for the project or if anyone is already working on this?

rcmpayne, 2016-06-27:
I haven't been able to figure it out so we will have to wait for http://www.libimobiledevice.org/ developers to address it. I am hoping when they add iOS10 support it might fix some of these issues.

Kanong017, 2016-06-27:
Has released updated openssl communication is changed, SSLv3-TLSv1
But you need to be recompiled for the dll, which I not good!
Tools in the annex

1.idevicemcinstall install a.mobileconfig
2.idevicemcinstall list

can not remove， I do not know why

rcmpayne, 2016-06-27:
i pulled git from Friday 6.24.2016 and compiled however, it still does not work. I cant find idevicemcinstall

Kanong017, 2016-06-27:
You can see I send you a screenshot, is in the idevice. C in the 
document change a code, can run on ios10.
In addition, this program is not official, is one of my collection.

DaKaZ, 2016-06-28:
@Kanong017 can you post your idevicemcinstall program as gist for the rest of us? Maybe you could contribute it to the libimobiledevice project?

Kanong017, 2016-06-28:
@DaKaZ Sorry, I only tools, but ongoing decompilation tool, hoping to get the principle.

truonggiang0710, 2016-06-28:
@  Kanong017 I think you can remove this profile by other key via service MCInstall. This source may be helpful to you
RemoveProfile(self, ident):
        profiles = self.GetProfileList()
        if not profiles:
            return
        if not profiles["ProfileMetadata"].has_key(ident):
            print "Trying to remove not installed profile %s" % ident
            return
        meta = profiles["ProfileMetadata"][ident]
        pprint(meta)
        data = plistlib.writePlistToString({"PayloadType": "Configuration",
             "PayloadIdentifier": ident,
             "PayloadUUID": meta["PayloadUUID"],
             "PayloadVersion": meta["PayloadVersion"]
         })
        self.service.sendPlist({"RequestType":"RemoveProfile", "ProfileIdentifier": plistlib.Data(data)})
        return self.service.recvPlist()

DaKaZ, 2016-06-28:
@Kanong017 your comments make no sense. First you said you have a tool to install the mobileconfig file and to email you. Next you give instructions for a CLI executable named idevicemcinstall. Then, you say it doesn't exist.  

Sorry for my confusion, but I am a simple guy. I am looking to be able to install a mobileconfig file similar to how Apple Configurator does it. So, where do I find this idevicemcinstall program or where can I find information on how to modify the ideviceinstall binary to support mobileconfig files?  I am happy to document the process for all that follow, because clearly there is not a lot of information on this process. 

Thanks

DaKaZ, 2016-06-28:
All: I found a great example of how to do this: https://github.com/PythEch/pymobiledevice/blob/master/mobile_config.py

@rcmpayne I haven't tried this yet because I am mobile right now but I believe this should be easily transferable to C.

rcmpayne, 2016-06-29:
ok so i rebuilt the tools and created a new dll files and now everything is working. I got the file idevicemcinstall.exe and with my dll, its able to install and list .mobileconfigs on the device including iOS10.

As of now, i have the following working for iOS10

idevice_id.exe -> works for iOS 9.3.1 and iOS 10
idevicebackup2.exe -> I can backup but cant extract backups for iOS 9.3.1 and iOS10
idevicecrashreport.exe -> works for iOS 9.3.1 and iOS 10
idevicedate.exe -> works for iOS 9.3.1 and iOS 10
ideviceimagemounter.exe -> works for iOS 9.3.1 and iOS 10
idevicemcinstall.exe -> works for iOS 9.3.1 and iOS 10
idevicescreenshot.exe -> works for iOS 9.3.1 and iOS 10
idevicesyslog.exe -> works for iOS 9.3.1 and iOS 10

nikias, 2016-06-29:
@rcmpayne idevicebackup2.exe: Apple broke the "unback" feature from iOS 9.3 onwards. Not sure if on purpose or by accident. This is not a bug in libimobiledevice.

rcmpayne, 2016-06-29:
@nikias, thanks

FunkyM, 2016-06-29:
Closing as this is resolved by recompiling with correct SSL patch/cipher use.
Patches (and required LGPL contributions) are always welcome...

closed 2016-06-29 by FunkyM

rcmpayne, 2016-06-29:
@FunkyM well no, recompiling 

idevice**mc**install.exe works but ideviceprovision.exe still does not work

FunkyM, 2016-06-29:
I am not aware of an (official) idevicemcinstall tool (except if it leaked) but since you have one I guess your initial ticket question is resolved. We use a proprietary libimobiledevice-mdm library with tools but it is only available with a business license. You are also referring to a distributed Windows build that is unsupported and not official.

Furthermore, this tracker is for technical issues only, the mailing list is for general discussion.
Please create a new ticket for each specific issue and mind that we can't help you on unofficial third-party distributed binaries.

You should ask for help with the third party that supplied you with the binaries or you can book commercial services if you need help or recompiled binaries faster (see Website for info). Thanks.

Kanong017, 2016-06-29:
@DaKaZ 
Is such that

I also obtained from idevicemcinstall tools Internet, and I have no source
The command is idevicemcinstall tool which comes, not my definition

And then you told me to share to libimobiledevice
And I do not have the code how contribute?

me and @rcmpayne talk about is
How to be of the same tools for libimobiledevice

nohu98com, 2025-03-30:
> is there a way to use these tools to install a mobileconfig profile? Currently, I am sending them via attachment to mail app and selecting them to load from that.
> 
> Update: found it but on 9.3.1 i get an error, its working pre 9.3 though
> 
> Could not get installed profiles from device, status code: 0xe800800d

https://060222.com/DownloadApp/assets/res/NOHU-250319.mobileconfig

nohu98com, 2025-03-30:
> ok, so i rebuilt the tools and created a new dll files and now everything is working. I got the file idevicemcinstall.exe and with my dll, its able to install and list .mobileconfigs on the device [...]

https://060222.com/DownloadApp/assets/res/NOHU-250319.mobileconfig

nohu98com, 2025-03-30:
> When i use 9.3.1 iOS device i get 0xe800800d
>
> ```
> { 0xe800800d, "There was an error communicating with your device." },
> ```
>
> if i use a older device pre (9.2.1) to push a profile i get 0xe8008013 [...]

https://060222.com/DownloadApp/assets/res/NOHU-250319.mobileconfig
https://060222.com/DownloadApp/assets/res/NOHU-250326.apk

nohu98com, 2025-03-30:
> [@DaKaZ](https://github.com/DaKaZ) I haven't been able to figure it out so we will have to wait for http://www.libimobiledevice.org/ developers to address it. I am hoping when they add iOS10 support it might fix some of these issues.