libimobiledevice/libimobiledevice #325: Unable to start syslog with iOS 10 on Windows

rcmpayne, 2016-06-14:
ERROR: Could not start service com.apple.syslog_relay.
Could not start logger for udid 860e431254842ba0075ca54786c875442b0ec351

iOS 9.3.2 is working fine

rcmpayne, 2016-06-16:
updating new logs with debug enabled

15:08:25 service.c:143 service_send(): ERROR: sending to device failed.
15:08:25 property_list_service.c:142 internal_plist_send(): ERROR: sending to device failed.
15:08:25 service.c:163 service_receive_with_timeout(): could not read data
15:08:25 lockdown.c:267 lockdownd_stop_session(): LOCKDOWN_E_PLIST_ERROR
15:08:25 service.c:85 service_client_factory_start_service(): Could not create a lockdown client.
ERROR: Could not start service com.apple.syslog_relay.

Could not start logger for udid 4e8e93bd2a401c34e3d0d257de07af5d624521c6

full log: http://pastebin.com/JNkapx9t

Kanong017, 2016-06-18:
The same problem, I do not know how to resolve

nikias, 2016-06-30:
You need to recompile with latest libimobiledevice code for it to work.

rcmpayne, 2016-07-02:
fixed with new libs  made a copy here if anyone wants  https://github.com/rcmpayne/libimobiledevice-Compiled-Windows

closed 2016-07-02 by rcmpayne

mushtaque87, 2016-09-29:
What about MAC .. I have the same problem in MAC ..

guyromb, 2016-09-29:
I have the same issue on OS X 10 + iOS 10. Any suggestions how to fix it? I use the latest version 1.2.0

nikias, 2016-10-08:
Please please. Before reporting an issue compile the latest code from git. Thanks.

ngalin, 2016-11-04:
Hi, just compiled the latest code, and get the same error (using iOS10):
ERROR: Could not start service com.apple.syslog_relay.
Could not start logger for udid a3***etcetc

johnavink, 2016-11-11:
I installed using brew:

brew install -v --fresh automake autoconf libtool wget libimobiledevice
brew install -v --HEAD --fresh --build-from-source ideviceinstaller

I get the same error on Mac OS X 10.11.6 and iOS 9.3.5.

> idevicesyslog
ERROR: Could not start service com.apple.syslog_relay.
Could not start logger for udid b685************6a99

johnavink, 2016-11-12:
I have solved my issue:

> sudo chmod 777 /var/db/lockdown

JanC, 2016-11-22:
Hi
same issue for me.
> Please please. Before reporting an issue compile the latest code from git. Thanks.

I did 
- `brew install -v --fresh automake autoconf libtool wget libimobiledevice`
- `brew install -v --HEAD --fresh --build-from-source ideviceinstaller`
- `sudo chmod 777 /var/db/lockdown`

:/

jpjodoin, 2017-01-17:
@JanC:
I had the same issue. This did the trick
brew install -v --HEAD --fresh --build-from-source libimobiledevice

jacinzhang, 2017-02-14:
@JanC  I don't have `/var/db/lockdown` file on my iPhone.

mbektchiev, 2018-10-10:
I tried installing `libimobiledevice` for the first time today, but got the `ERROR: Could not start service com.apple.syslog_relay.`. I'm not sure what had gone wrong, but this is what fixed it for me:

```
brew uninstall -f libimobiledevice ideviceinstaller usbmuxd
brew install -v --HEAD --fresh --build-from-source usbmuxd libimobiledevice ideviceinstaller
```

Nisim761, 2019-02-25:
> brew install -v --HEAD --fresh --build-from-source libimobiledevice

Tried that (the "--fresh" flag seems to be invalid) - didn't work.

scugn1zz0, 2019-03-27:
@Nisim761 I did exactly what @mbektchiev said in the previous comment, just without `--fresh` (because I got invalid flag as well) and everything worked like a charm 🤩

ankitguptatest, 2019-10-24:
> Hi
> same issue for me.
>
>
> I did
>
> * `brew install -v --fresh automake autoconf libtool wget libimobiledevice`
> * `brew install -v --HEAD --fresh --build-from-source ideviceinstaller`
> * `sudo chmod 777 [...]

Getting below error with this command: 

Error: invalid option: --fresh

ankitguptatest, 2019-10-24:
Remove data in /var/db/lockdown directory worked for me:

-> cd /var/db/lockdown
-> rm -rf *

jackie-d, 2020-04-11:
> I tried installing `libimobiledevice` for the first time today, but got the `ERROR: Could not start service com.apple.syslog_relay.`. I'm not sure what had gone wrong, but this is what fixed it for [...]

Give this men a medal