libimobiledevice/libimobiledevice #327: It does not support ios10, lockdown error -5, how do?

Kanong017, 2016-06-18:
It does not support ios10, lockdown error -5, how do?

zgoda-mobica, 2016-06-20:
-5 is SSL error, is your binary compiled with openssl or gnutls?

Kanong017, 2016-06-20:
In addition ios10 device, other systems do not have this problem

rcmpayne, 2016-06-20:
So it depends on what app you are running. I am compiling with mingw using ssl. as @Kanong017 Kanong017 advised it works for all older devices including 9.3.2

```
admin@DEV /c/libimobiledevice
$ openssl
OpenSSL> version
OpenSSL 1.0.2h  3 May 2016
```

```
./configure --without-cython 
Make
Make install
```

```
C:\idevice_id.exe
ERROR: Could not connect to lockdownd, error code -5
```

```
C:\idevicesyslog.exe
ERROR: Could not start service com.apple.syslog_relay.
Could not start logger for udid 4e8e93bd2a4*********24521c6
```

Kanong017, 2016-06-20:
@rcmpayne yes,It does not support ios10, but is supported iTunesMobileDevice

meirtsvi, 2016-06-21:
can you further explain the last comment? does idevicescreenshot uses iTunesMobileDevice?
is there a way to take snapshot on iOS 10 device?

Kanong017, 2016-06-21:
After testing, the screenshot does not support ios10 in the iTunesMobileDevice

nikias, 2016-06-21:
Make sure to recompile using latest git head. iOS 10 devices don't allow SSLv3 anymore but require at least TLSv1.

meirtsvi, 2016-06-21:
Worked like a charm, thanks!

closed 2016-06-21 by FunkyM

mnang, 2016-09-28:
I've similar issue in iOS 10.
what do you mean by ecompile using latest git head.

Thanks

nikias, 2016-09-28:
@mnang check out the latest code from git and compile it.

FredMour, 2016-12-27:
could you release this fix in a release version ?