libimobiledevice/libimobiledevice #330: Can skip set up activation via libimobiledevice ?

truonggiang0710, 2016-06-27:
I have an iPhone 5 that just restore to 9.3.2.
I'm very lazy. So I want to active my iPhone with skip all set up ( same as Apple configurator 2 ). 
Can libimobiledevice do that ?
Please help me when you have free time!

Thanks a lot !

truonggiang0710, 2016-07-11:
Can anybody help me, please ? I just want to skip Language screen and Region screen.
Please drop me some information when you're free.
Thanks a lot !

FunkyM, 2016-07-12:
No, libimobiledevice can not currently do it although it is possible. I also doubt that you are lazy.

closed 2016-07-12 by FunkyM

truonggiang0710, 2016-07-13:
Hi ! In some case, laziness will make the world better !
Thanks a lot for your response !

iOSShohrab, 2016-12-07:
@truonggiang0710 did you able to get any solution for this?

truonggiang0710, 2017-05-29:
@iOSShohrab Yes, I find out !

iamluosm, 2017-05-31:
Hi man, how did you do that ?

truonggiang0710, 2017-05-31:
@iamluosm sorry. But I don't have permisson to show it for you here. Please contact to me over email dotruonggiang88@gmail.com

Roger-mazheng, 2019-09-03:
hi man,how did you du that? can you tell me?  i need the solution.

yunhailu, 2023-03-31:
> hi man,how did you du that? can you tell me? i need the solution.

Have you solved this problem?

mroger-tech, 2023-04-03:
> > hi man,how did you du that? can you tell me? i need the solution.
> 
> Have you solved this problem?

Yes,i did it,try the following code.

lockdownd_client_t lockdown = NULL;
	lockdownd_error_t ret = LOCKDOWN_E_UNKNOWN_ERROR;

	if (LOCKDOWN_E_SUCCESS != (ret = lockdownd_client_new_with_handshake(device, &lockdown, "skipsetup"))) {
		printf("ERROR: Could not connect to lockdownd, error code %d\n", ret);
		result = -3;
		goto cleanup;
	}

	if (lockdownd_set_value(lockdown, "com.apple.international", "Locale", plist_new_string("zh_CN")) == LOCKDOWN_E_SUCCESS) {
		printf("set Locale [zh_CN]\n");
	}
	if (lockdownd_set_value(lockdown, "com.apple.international", "Language", plist_new_string("zh-Hans")) == LOCKDOWN_E_SUCCESS) {
		printf("set Language [zh-Hans]\n");
	}
	if (lockdownd_set_value(lockdown, "com.apple.purplebuddy", "SetupDone", plist_new_bool(1)) == LOCKDOWN_E_SUCCESS) {
		printf("set SetupDone [true]\n");
	}
	if (lockdownd_set_value(lockdown, "com.apple.purplebuddy", "SetupFinishedAllSteps", plist_new_bool(1)) == LOCKDOWN_E_SUCCESS) {
		printf("set SetupFinishedAllSteps [true]\n");
	}
	if (lockdownd_set_value(lockdown, "com.apple.purplebuddy", "ForceNoBuddy", plist_new_bool(1)) == LOCKDOWN_E_SUCCESS) {
		printf("set ForceNoBuddy [true]\n");
	}
	else {
		printf("ERROR:set ForceNoBuddy failed\n");
		result = -4;
		goto cleanup;
	}

Lbywer, 2023-08-09:
> > > > hi man,how did you du that? can you tell me? i need the solution.
> > > 
> > > 
> > > Have you solved this problem?
> > 
> > 
> > Yes,i did it,try the following code.
> > lockdownd_client_t lockdown = NULL; lockdownd_error_t ret = LOCKDOWN_E_UNKNOWN_ERROR;
> > ```
> > if (LOCKDOWN_E_SUCCESS != (ret = lockdownd_client_new_with_handshake(device, &lockdown, "skipsetup"))) {
> > 	printf("ERROR: Could not connect to lockdownd, error code %d\n", ret);
> > 	result = -3;
> > 	goto cleanup;
> > }
> > 
> > if (lockdownd_set_value(lockdown, "com.apple.international", "Locale", plist_new_string("zh_CN")) == LOCKDOWN_E_SUCCESS) {
> > 	printf("set Locale [zh_CN]\n");
> > }
> > if (lockdownd_set_value(lockdown, "com.apple.international", "Language", plist_new_string("zh-Hans")) == LOCKDOWN_E_SUCCESS) {
> > 	printf("set Language [zh-Hans]\n");
> > }
> > if (lockdownd_set_value(lockdown, "com.apple.purplebuddy", "SetupDone", plist_new_bool(1)) == LOCKDOWN_E_SUCCESS) {
> > 	printf("set SetupDone [true]\n");
> > }
> > if (lockdownd_set_value(lockdown, "com.apple.purplebuddy", "SetupFinishedAllSteps", plist_new_bool(1)) == LOCKDOWN_E_SUCCESS) {
> > 	printf("set SetupFinishedAllSteps [true]\n");
> > }
> > if (lockdownd_set_value(lockdown, "com.apple.purplebuddy", "ForceNoBuddy", plist_new_bool(1)) == LOCKDOWN_E_SUCCESS) {
> > 	printf("set ForceNoBuddy [true]\n");
> > }
> > else {
> > 	printf("ERROR:set ForceNoBuddy failed\n");
> > 	result = -4;
> > 	goto cleanup;
> > }
> > ```
> 
> Hello @mroger-tech , does this make sense for iOS16.4+ devices ? As far as checked - no, may be you have some ideas about how to bypass Setup app on latest iOS versions?

hello, when i try this for iOS 14+ devices? its failed. do you have any idea with this. best glad.

ScottPierce, 2023-10-10:
@mroger-tech Have you gotten this working for iphone 16+?