libimobiledevice/libimobiledevice #333: Question: trigger sysdiagnose using libimobiledevice

rcmpayne, 2016-06-30:
does the tools have the ability to trigger a sysdiagnose log dump (Volume Up + Down, and Power button simultaneously)?

nikias, 2016-06-30:
Oh I didn't know about this. Where will the log file be stored? This way it might be possible to track back to where it gets initiated from. There's a diagnostics_relay service but I don't think it has a command for that.

rcmpayne, 2016-07-02:
not sure where its stored but will find out for you

rcmpayne, 2016-11-10:
They are synced with iTunes or youcan pull them using the imobiledevice

iTUnes: C:\Users\<user>\AppData\Roaming\Apple Computer\Logs\CrashReporter\MobileDevice\Roger Calvin M's iPhone\DiagnosticLogs\sysdiagnose\sysdiagnose_2016.11.10_12-04-26-0400

disks.txt
error_log.txt
ps.txt
ps_thread.txt
spindump.txt
spindump-nosymbols.txt
sysdiagnose.log
tailspin-info.txt
taskinfo.txt
vm_stat.txt
zprint.txt

rcmpayne, 2017-01-11:
here are the ios device side paths

Copy: /Baseband/log-bb-2016-11-28-stats.plist
Copy: /Baseband/log-bb-2016-11-30-stats.plist
Copy: /Baseband/log-bb-2016-12-02-stats.plist
Link: /DiagnosticLogs/sysdiagnose/Latest
Unable to open device file './DiagnosticLogs/sysdiagnose/Latest' (10). Skipping...
Copy: /DiagnosticLogs/sysdiagnose/sysdiagnose_2017.01.11_15-21-02-0400.tar.gz

rcmpayne, 2018-11-12:
bump

any updates on if this is possible?

Dantee296, 2021-07-05:
@rcmpayne did you find any way to execute it remotely ?

TaihouKai, 2024-01-23:
Any updates since then?