libimobiledevice/libimobiledevice #338: idevicescreenshot produces invalid tiff on iOS9

progressnerd, 2016-07-07:
Taking a screenshot of an iOS 9 device (on a few iOS 9.3 devices I tried, specifically) produces a TIFF with an invalid header. My mac opens it fine, but the default image viewer on Ubuntu does not, and my attempts to covert the image to a different format with various image processing utilities failed. They all complained that the two byte order bytes were neither 0x4949 ("II") or 0x4d4d ("MM") as expected. This does not occur with any of my iOS 8 devices.

progressnerd, 2016-07-07:
Oh, it appears that the screenshot is a png, not a tiff. That's fine by me. It just means the documentation, help text, and default file extension should change for these newer versions.

nikias, 2017-06-18:
Closing, fixed with commit 835d84b678d23f92622445f5bf142c2bf52a7e9b.

closed 2017-06-18 by nikias