libimobiledevice/libimobiledevice #343: Applications dictionary missing in info.plist during mobilebackup2 backup session

NitzanDavari, 2016-07-14:
I noticed mobilebackup_factory_info_plist_new doesn't create Applications dictionary as iTunes does.
As result the user's apps aren't downloaded automatically in the restored device. 
I noticed the dictionary fills each app with the following information:
app id , ApplicationSINF , PlaceholderIcon , iTunesMetadata.
I do know how to retireve app ids using installation proxy, however the i failed to retrieve the other three ( attempted to use instproxy_lookup but failed - any ideas solution for this

mexmer, 2016-07-14:
ApplicationSINF and iTunesMetadata can be retrieved from installation proxy, as for PlaceholderIcon, i don't think i have seen that attribute there.

NitzanDavari, 2016-07-17:
@mexmer  thank you for your reply.
first question, is how do I retrieve ApplicationSINF and iTunesMetadata can be retrieved from installation proxy - i tried with iDeviceInstaller, and i couldn't trace that information - here is the output I received 
[out.txt](https://github.com/libimobiledevice/libimobiledevice/files/367688/out.txt)

And regarding to your second comment - it is related to the springboard - according to my testings, if this dictionary is available in the info.plist the restored device loads with the source device apps like this:
[image]
and then download the apps to the restored device. When that dictionary isn't available no buffered\placeholders apps are seen in the target device.

Once again thank you for your support!

mexmer, 2016-07-17:
as for 1st question, i'm not sure, if iDeviceInstaller can or cannot do that, i don't use it. i have my own code, that calls installation_proxy service.
you can specify, which attributes you want in proxy listing output.

as for placeholder icons, if you don't specify them, you just see "generic" download icon, so imo it's not big issue, otherwise you will need to use springboard service, to retrieve app icon, and put it inside that block. although on iOS 9 devices might be possible to retrieve placeholder icon from install_proxy, i have not tested that.

NitzanDavari, 2016-07-17:
Many thanks, i'll give it a try.
Your assistance is highly appreciated :)

mexmer, 2016-07-18:
just checked some of my old backups, and Applications dictionary is not there (and i mean backups created by iTunes, not idevicebackup/idevicebackup2).

Most likely this was added by recent version of iTunes, since iTunes versions found i my backups were
9.0.3, 10.1.0.1, 11.1.2, 12.1.1 

i can see Applications dictionary only in backup from latest version of iTunes 12.4.1.6, it's possible it was added in any version starting 12.2 (since i don't have backup from that version). Which was first version supporting iOS 9 i think.

mexmer, 2016-07-18:
also one more note, it doesn't add "Application" dictionary for pres iOS 9.
i suspect it might have something with new handling of application packages in iOS 9, while on elder iOSes was 1 universal package for all devices, since iOS 9, there are multiple packages deploying on specific devices.

NitzanDavari, 2016-07-18:
@mexmer  further explanation for the absence of applications dictionary to pre iOS 9 devices - app archiving was blocked for iOS 9+, so apparently this is the new approach for apps transfer\restoration

mexmer, 2016-07-18:
@NitzanDavari thanks for comment, i'm aware of removal of archiving function in iOS9.
But you are right, it also might be related. 

Atm. i'm not aware how iTunes utilizes informations in Application dictionary tho'

daniel-bluesea, 2016-09-15:
@mexmer @NitzanDavari so the bottom line is that from ios9 you cant backup & restore applications using the idevicebackup2?

do you know if the ideviceinstaller can be used to archive and install applications?

thanks

mexmer, 2016-09-15:
no, it cannot be used for archiving aplications or their data from iOS 9 onwards due removing archive functioniality. and due fundamental changes in application packaging, even installing packages from archive IPA might be problematic, but i had some success installing application archived from ios 8 to ios 9 device.

daniel-bluesea, 2016-09-18:
@mexmer thank you for your answer.
so how can i perform a restore of application to a new device as iTunes does it?
maybe using afc2?
until now, libimobiledevice project manage to replace the need to use iTunes, except this application restore issue..

daniel-bluesea, 2016-09-19:
@NitzanDavari any luck in building the info.plist so applications are restored?

daniel-bluesea, 2016-09-19:
@NitzanDavari even when i'm using iTunes generated Info.plist file with idevicebackup2 restore, i dont see the applications waiting icons...

mexmer, 2016-09-19:
@daniel-bluesea idevicebackup can't restore application. that functionality is not implemented afaik.
while device reads info.plist during restore operation, i believe it's itunes who does purchase restore.

i'm not sure how is it done tho'.

i have my own code based on libimobile communcation with backup/restore service, and it doesn't recover apps/icons either.

on elder ios(prior ios 9) i used archive function to pull apk from device, then install to restore that apk. but now i don't know how is it done.

but i know, that iTunes can restore device from backup generated by my app, including apps (after i fixed info.plist).

daniel-bluesea, 2016-09-22:
@mexmer the part when iTunes adds the restored application to the target device, is the part that is interesting to me..

maybe do you know if somehow you can call iTunes to restore (including apps) from the command line?

calling AppleMobileBackup.exe -r only do a "regular" restore, without the applications..

mexmer, 2016-09-22:
that's what i'm trying to figure out. and yes, i'm aware that their console tool also doesn't restore apps. that's why i believe, it's something that iTunes does.
Also have not figured, if it can restore from "any" backup, or just own backup and if you need/or don't need to transfer purchases to computer that restore apps.

nikias, 2016-09-23:
I just commited https://github.com/libimobiledevice/libimobiledevice/commit/00424f40574641d1360c9b1115770f5cc5d19a72 which will add the ```Applications``` dictionary and the ```Installed Applications``` array to the Info.plist. Feel free to test and report back.

nikias, 2016-09-23:
What I noticed on one of my devices is that iTunes apparently created the ```/var/mobile/Media/iTunesRestore``` directory with a file ```RestoreApplications.plist``` that contains exactly the same info found in the ```Applications``` dictionary of the Info.plist. I didn't test if the backup restore would actually create this file for us or if iTunes creates it.

nikias, 2016-09-24:
Ok I can confirm that the backup restore itself is not triggering the installation of these apps when the info is present. Also creating the mentioned file doesn't do anything. But the Applications dictionary information will be used in conjunction with installation proxy to put "placeholders" on the screen for the apps with the contained icon. I need to check how to trigger the installation; I suppose it's possible by sending the ApplicationSINF and iTunesMetadata to the device without having the actual .ipa file. The device will then download the apps.

mexmer, 2016-09-25:
@nikias seems we thinking along same lines, i was not succesfull yet with invoking correct install proxy command to achieve such thing so. i'm putting sinf, itunesmedatadata and prerendered icon into options dictionary.

as for retrieving application icon (to store it into info.plist) i retrieve that one from springboard service (eg. using regular application icon), might be not best, but imo sufficient.

as for Dictionary itself (in info.plist), it's creation seems to have started with itunes 12.2.1 or later, and is done only for ios9 and later (i suspect it's because for previou ioses you could have restored application using IPA files, which were downloaded by itunes, when you did purchase transfer, or during sync)

when i have chance, i will probly try to use hardware USB monitor, to capture restore process, and see if applications are restored and how it's done ... not sure when that will be, tho'

daniel-bluesea, 2016-09-25:
@nikias cool!
yes, the file itself RestoreApplications.plist doesn't trigger anything, but it is used when the restore is done.
i think that itunes (via installation_proxy??) actually download from app store the applications listed at RestoreApplications.plist.. the "placeholder" are actual app icons waiting for internet connection to be downloaded.. so i think its not really a placeholder but actual download as you can do from the app store yourself..

anyway, i'll look into ideviceinstaller.c to try to understand how to trigger installation based on ApplicationSINF and iTunesMetadata ..

and of course wait to your input/solution

daniel-bluesea, 2016-09-26:
@mexmer regarding USB monitor:
it seems that iTunes does some "purplebuddy" queries, also some queries to com.apple.mobile.data_sync.domain_changed service, some home screen icons metrics queries, some capabilities queries and creates/call SyncAfterRestore at iTunes_control/iTunes

nikias, 2016-09-26:
@daniel-bluesea @mexmer if you look at my recent commit you will see that I get the PlaceholderIcon images from springboardservices. That's what iTunes does too, I dumped the communication.
Regarding the restore; iTunes will install .ipa files if you have them locally in your iTunes Library (even if they are outdated!). But if you don't have them (and I renamed the "Mobile Applications" folder) iTunes will tell the device to install the apps. I don't think iTunes is downloading them because then it would store them somewhere (ideally, in the Mobile Applications folder). But I am not 100% sure. In case you are able to get a USB dump during the restore (i.e. after the device reboot), hit me up on IRC/freenode user: nikias - I am happy to take a look.

daniel-bluesea, 2016-09-26:
@nikias

[[2016-09-26 session 181857] Trace0001.zip](https://github.com/libimobiledevice/libimobiledevice/files/493471/2016-09-26.session.181857.Trace0001.zip)

bplist00...."#_..DLMessageDownloadFiles...

daniel-bluesea, 2016-09-26:
[[2016-09-26 session 185625] Trace0000.zip](https://github.com/libimobiledevice/libimobiledevice/files/493615/2016-09-26.session.185625.Trace0000.zip)

daniel-bluesea, 2016-09-26:
since ios 9, there is no app archive.. so i dont think they save it on the pc..
somehow they place a placeholder icon that upon network connection start download the app (from app store?? icloud??), but i could be wrong..

daniel-bluesea, 2016-09-27:
@nikias the usb dumps were any good?

nikias, 2016-09-27:
@daniel-bluesea didn't have a chance to take a look yet. Will do today and report back what I found.

mexmer, 2016-09-27:
their viewer seems to missing full text search in data, which makes it hard to explore, will need some time to look at that.

daniel-bluesea, 2016-09-27:
@mexmer can you refer me to a usb monitor that has full text search?

daniel-bluesea, 2016-09-27:
@nikias itunes calls AppleMobileBackup.exe with pipe, attached the traffic between them, (see AMSRestoreWithApplicationsRequest)

[itunes-restore.txt](https://github.com/libimobiledevice/libimobiledevice/files/495568/itunes-restore.txt)

mexmer, 2016-09-27:
i use following device in work
http://www.totalphase.com/products/beagle-usb480/
datacenter software (which is monitor and viewer) has full text search function. as for protocol filters and decoders, it's quite lacking them, but they have api documentation, so you can write your own.

mexmer, 2016-09-27:
datacenter is free for download (you just need to register account with them)
there is only shazam and one game installed on restored phone
com.pixelfederation.diggy
com.shazam.Shazam

[apple restore.zip](https://github.com/libimobiledevice/libimobiledevice/files/495988/apple.restore.zip)

daniel-bluesea, 2016-09-27:
@mexmer  apple_restore.zip is your usb dump?

mexmer, 2016-09-28:
yes, it's communication between phone and windows version of itunes, after phones was reset from settings.

daniel-bluesea, 2016-09-28:
@nikias anything else you need?

daniel-bluesea, 2016-09-28:
@mexmer what do you mean "after phones was reset from settings"?

i think we need usb dump for when the phone is rebooted after restore..

mexmer, 2016-10-03:
looks like in 1st phase /iTunesRestore directory containing RestoreApplications.plist is created
RestoreApplications.plist contains application dictionary.

have not fully tested it yet tho'

mexmer, 2016-10-04:
i can confirm it somewhat working, not sure what problem is, or what files i'm missing, but sometimes instead of app icon i get "generic" application icon, but all apps are loading.

write /iTunesRestore/RestoreApplications.plist
invoke restore
after phone restart (done automaticaly) and finishing touches, apps are downloading.

tested on iPad Air with ios 9.3.4 and iPhone 5s with ios 9.3.5

daniel-bluesea, 2016-10-05:
@mexmer "invoke restore": do you mean iTunes restore or idevicebackup2 restore?

mexmer, 2016-10-05:
i mean restore operation
eg. start backup service with restore command

this functionality can be added to idevicebackup2, it's aditional step prior calling backup service.

nikias, 2016-10-05:
@mexmer hey cool. I will add this feature to the restore operation. Maybe I should add a command line switch for it.

mexmer, 2016-10-05:
i think you can.
but if you add switch, maybe it should also report, if Application dictionary is missing from restored backup ... therefore it's not possible to restore apps

mexmer, 2016-10-05:
btw. @nikias /var/mobile/Media/ is root of what you get, when connecting to com.apple.afc service :D
i forgot to write that, when replying some days ago to RestoreApplications.plist note.

what apps i'm working on does is
start notification proxy, send "restore in progress" notification.
write restoreapplication.plist
start backup service
do restore

phone restarts automaticaly (i think it does twice)
i did disconnect phone from computer after 1st restart

after second restart you get welcome screen "your phone has been update", set touchid/passcode, and then you can check if apps are restoring ... they should. i tested that with google maps and skymap (those are huge, few hundreds megabytes)

if you have password protected backup, it will restore wifi settings including passwords, so it did automatically connect to our wifi, and downloaded everything.

mexmer, 2016-10-05:
one more thing, i have not tested (since unlike app restore, there is no interest here in that)
there are multiple "itunes" related files in info.plist, that (From log i posted above), seem are restore before restore starts 
from log it looks like
itunes post sync notification (trough np)
itunes create /itunes-sync.lock file
itunes restore files like ibooks.plist and so on
itunes create apprestore list
itunes remove /itunes-sync.lock
itunes send syncstop notification (trough np)
backup service is started and restore proceeds

nikias, 2016-10-05:
Hm. Well we either do it right or don't do it at all :) The purpose of idevicebackup is to mimic whatever iTunes is doing. I thought the restore would take care of iBooks*.plist though. Need to check this.

mexmer, 2016-10-05:
@nikias check log i posted above, you can see, that backup operation starts later, there is lot of itunes file copy stuff before that.

i think around index 12475 starts real backup - there is start session, which starts com.apple.mobilebackup2 (that part is encrypted)

mind this is whole communication, when you connect factory reset phone (or new unpacked phone) to itunes and it's only showing with welcome (and proceed to configuration)

mexmer, 2016-12-02:
@nikias maybe you should put restore onto milestone list, for version 1.5 or so 😺

skortela, 2017-03-16:
@nikias, @mexmer Can the idevicebackup2 be fixed to support restoring apps?

corsac-s, 2018-04-18:
Hey,
I just restored an iPhone backup using libimobiledevice git (3a37a4e). Apps are not restored. Looking at least few messages from @mexmer I had the feeling that some change had been committed adding that support, but I'm unsure now. Any pointers on this? Thanks!

mexmer, 2018-04-18:
you can check sourcecode, but i don't think it was added to idevicebackup2

samuelsadok, 2018-05-13:
The discussion has become a bit long and hints are scattered and not very clear for non-contributors so let me summarize how to manually workaround the issue.

1. Locate Info.plist in your backup: `backup-dir/[old device udid]/Info.plist`. At the top it contains an `Applications` dictionary:
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Applications</key>
        <dict>
                <key>com.Hamway.EMFMeter</key>
                <dict>
                        <key>ApplicationSINF</key>
                        <data>
                        AAAEIHNpbmYAAAAMZnJtYWdhbWUAAAAUc2NobQAAAABpdHVuAAAA
                        AAAAA3BzY2hpAAAADHVzZXJRzxOcAAAADGNyZHTWOlqGAAAADGFz
                        ZHQAAAAAAAAADGtleSAAAAAFAAAAGGl2aXaqx1aqkhQpAEbdqoqz
                        M8M+AAAAWHJpZ2h2ZUlEAABxBHBsYXQAAAACYXZlcgEBAQB0cmFu
[...]
```

2. Mount the device to be restored. I use `ifuse ~/mnt/MyPhone`
3. On the device, create a file `iTunesRestore/RestoreApplications.plist`. (in my case this would be `~/mnt/MyPhone/iTunesRestore/RestoreApplications.plist`), containing the contents of the `Application` dictionary from Info.plist. So it will look something like this:

```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>com.Hamway.EMFMeter</key>
        <dict>
                <key>ApplicationSINF</key>
                <data>
                AAAEIHNpbmYAAAAMZnJtYWdhbWUAAAAUc2NobQAAAABpdHVuAAAA
                AAAAA3BzY2hpAAAADHVzZXJRzxOcAAAADGNyZHTWOlqGAAAADGFz
                ZHQAAAAAAAAADGtleSAAAAAFAAAAGGl2aXaqx1aqkhQpAEbdqoqz
                M8M+AAAAWHJpZ2h2ZUlEAABxBHBsYXQAAAACYXZlcgEBAQB0cmFu
[very long]
                MwI7AkYCTgJQAlUChgKLAqUCpgLhAuYC7ALxAvQC9gMDAxgDGwMg
                AyEDLgMvAzkDOgAAAAAAAAIBAAAAAAAAAD8AAAAAAAAAAAAAAAAA
                AANb
                </data>
        </dict>
</dict>
</plist>
```

4. Start the restore. I used `idevicebackup2 --source [old device udid] --interactive restore --system --reboot --copy --settings [backup-dir]`. After the reboot, once I got to the home screen the applications were be there and started to download. App data was preserved as expected (WhatsApp messages, game statistics, ...).

nikias, 2018-05-13:
@samuelsadok thanks for the conclusion and test results. I had the same observationd but never tested this. Time for me to finally integrate this into the tool itself!

corsac-s, 2018-05-14:
@samuelsadok thanks for the method. Next time I have to restore a backup (gasp) I'll try to test it.

samuelsadok, 2018-05-14:
@mexmer Yes true, but as someone not in the iOS hacker scene some details were unclear to me. For instance I thought the path `/iTunesRestore/RestoreApplications.plist` was with respect to the iOS root (which I know is locked down) rather than the directory mounted by ifuse (aka AFC root). As a consequence I was surprised to find that the issue could be solved without modifying source code.

mexmer, 2018-05-14:
@samuelsadok since you connect to AFC service, you are restricted to sandboxed folder. itunes will literally need to hack iOS to get access to root folder :smile_cat:

nikias, 2018-05-20:
@mexmer I went through your dump and while I can see that it does a number of AFC operations including creating a file ```/iTunes_Control/iTunes/iTunesLock``` that is deleted after completion of the restore, there is no PostNotification. Anyway, we have the writing of the 'iTunes Files' missing that I will add.
@samuelsadok thanks for the pull request but I think it is not 100% accurate since it seems like iTunes doesn't do the PostNotification and also it is using a different lock file.
Just be patient a little bit, I will add the necessary changes.

samuelsadok, 2018-05-21:
@nikias Ah yeah I didn't focus much on accuracy, I just wanted to quickly fix the issue for myself and pushed it in case someone else wants to use it.
Feel free to edit the PR or do a different implementation altogether if you think that's better.

mexmer, 2018-05-21:
@nikias as for notifications, you are right, i thought it's posting notificication, but it only register observers.
but as for lock files, i think file you mentioned is related to restoring itunes data, not application.

what is important is to create
/com.apple.iTunes.restore-lock
which is general lock file for whole process.

although they might be inclusive, so at some point you need to have both of the.

nikias, 2018-05-21:
@mexmer ```/com.apple.iTunes.restore-lock``` is only tested for existence (AFC_OP_GET_FILE_INFO == 0xA), and the result code is ENOENT (8). It is not created. Maybe it is deleted when found, but I haven't tested that.

[3 more replies]