libimobiledevice/libimobiledevice #346: libimobile is not working with iOS 10.0 beta version(Could not connect to lockdownd, error code -5)

AbhilashMathur, 2016-07-20:
I am using libimobiledevice in my iOS project. I upgrade my iphone 6sPlus to iOS 10.0 beta version. i fired ideviceinfo command from my mac and get the response : Could not connect to lockdownd, error code -5

Any one know about this error??

when i am trying to install my ipa through ideviceinstaller, there I am having the following error: Could not connect to lockdownd. Exiting.

please let me know if someone know's the solution!

Kanong017, 2016-07-20:
ssh mode changed
old:SSLv3
new:TLSv3

mexmer, 2016-07-20:
guess you mean ssl mode, also it's not TLSv3 but TLSv1.0

AbhilashMathur, 2016-07-20:
@Kanong017  can you please elaborate your answer? I am not getting your answer?

mexmer, 2016-07-20:
@AbhilashMathur latest libimobile supports iOS10 devices. it supports them since change from SSLv3_method(which is strict SSLv3) to SSLv23_method (which does autonegotiate from TLSv1.2 down to SSLv2). @FunkyM  might tell you exactly what version it was.

mexmer, 2016-07-20:
@Kanong017  That one is valid only if compiled with openssl 1.0.2 and later, with openssl 1.0.1 and lower you need to use sslv23_method, since tlsv1 is strictly TLSv1.0

AbhilashMathur, 2016-07-21:
@Kanong017 @mexmer i downloaded openssl 1.0.2, but still same error occurs while firing ideviceinfo : Could not connect to lockdownd, error code -5.

i checked my TLSv1_method in idevice.c file, same method was presented in the file.
 
i have latest package of libimobiledevice 1.2.0, still its is not working!.

what wrong i am doing??

rcmpayne, 2016-07-21:
i just compiled and its working for me @AbhilashMathur you can download and check with my dll's here https://github.com/rcmpayne/libimobiledevice-Compiled-Windows

AbMathur, 2016-07-22:
@rcmpayne m working on ubuntu 14.04, can you please help in that?

AbMathur, 2016-07-22:
@Kanong017 @mexmer  i downloaded openssl 1.0.2 on ubuntu 14.04 & start compling all packages. I compiled all latest packages in this sequence :

libxml
libusb 
fuse
libplist
libusbmuxd
libimobiledevice
usbmuxd
ideviceinstaller
ifuse

while compiling ideviceinstaller packges i stuck with this error:

make  all-recursive
make[1]: Entering directory `/home/administrator/Desktop/libimobile_latest/ideviceinstaller-1.1.0'
Making all in src
make[2]: Entering directory `/home/administrator/Desktop/libimobile_latest/ideviceinstaller-1.1.0/src'
  CC       ideviceinstaller-ideviceinstaller.o
ideviceinstaller.c: In function ‘main’:
ideviceinstaller.c:1058:4: error: passing argument 4 of ‘instproxy_install’ from incompatible pointer type [-Werror]
    instproxy_install(ipc, pkgname, client_opts, status_cb, NULL);
    ^
In file included from ideviceinstaller.c:45:0:
/usr/local/include/libimobiledevice/installation_proxy.h:234:19: note: expected ‘instproxy_status_cb_t’ but argument is of type ‘void (*)(const char *, void *, void *)’
 instproxy_error_t instproxy_install(instproxy_client_t client, const char *pkg_path, plist_t client_options, instproxy_status_cb_t status_cb, void *user_data);
                   ^
ideviceinstaller.c:1065:4: error: passing argument 4 of ‘instproxy_upgrade’ from incompatible pointer type [-Werror]
    instproxy_upgrade(ipc, pkgname, client_opts, status_cb, NULL);
    ^
In file included from ideviceinstaller.c:45:0:
/usr/local/include/libimobiledevice/installation_proxy.h:262:19: note: expected ‘instproxy_status_cb_t’ but argument is of type ‘void (*)(const char *, void *, void *)’
 instproxy_error_t instproxy_upgrade(instproxy_client_t client, const char *pkg_path, plist_t client_options, instproxy_status_cb_t status_cb, void *user_data);
                   ^
ideviceinstaller.c:1077:3: error: passing argument 4 of ‘instproxy_uninstall’ from incompatible pointer type [-Werror]
   instproxy_uninstall(ipc, appid, NULL, status_cb, NULL);
   ^
In file included from ideviceinstaller.c:45:0:
/usr/local/include/libimobiledevice/installation_proxy.h:283:19: note: expected ‘instproxy_status_cb_t’ but argument is of type ‘void (*)(const char *, void *, void *)’
 instproxy_error_t instproxy_uninstall(instproxy_client_t client, const char *appid, plist_t client_options, instproxy_status_cb_t status_cb, void *user_data);
                   ^
ideviceinstaller.c:1253:3: error: passing argument 4 of ‘instproxy_archive’ from incompatible pointer type [-Werror]
   instproxy_archive(ipc, appid, client_opts, status_cb, NULL);
   ^
In file included from ideviceinstaller.c:45:0:
/usr/local/include/libimobiledevice/installation_proxy.h:325:19: note: expected ‘instproxy_status_cb_t’ but argument is of type ‘void (*)(const char *, void *, void *)’
 instproxy_error_t instproxy_archive(instproxy_client_t client, const char *appid, plist_t client_options, instproxy_status_cb_t status_cb, void *user_data);
                   ^
ideviceinstaller.c:1391:3: error: passing argument 4 of ‘instproxy_restore’ from incompatible pointer type [-Werror]
   instproxy_restore(ipc, appid, NULL, status_cb, NULL);
   ^
In file included from ideviceinstaller.c:45:0:
/usr/local/include/libimobiledevice/installation_proxy.h:349:19: note: expected ‘instproxy_status_cb_t’ but argument is of type ‘void (*)(const char *, void *, void *)’
 instproxy_error_t instproxy_restore(instproxy_client_t client, const char *appid, plist_t client_options, instproxy_status_cb_t status_cb, void *user_data);
                   ^
ideviceinstaller.c:1399:3: error: passing argument 4 of ‘instproxy_remove_archive’ from incompatible pointer type [-Werror]
   instproxy_remove_archive(ipc, appid, NULL, status_cb, NULL);
   ^
In file included from ideviceinstaller.c:45:0:
/usr/local/include/libimobiledevice/installation_proxy.h:372:19: note: expected ‘instproxy_status_cb_t’ but argument is of type ‘void (*)(const char *, void *, void *)’
 instproxy_error_t instproxy_remove_archive(instproxy_client_t client, const char *appid, plist_t client_options, instproxy_status_cb_t status_cb, void *user_data);
                   ^
cc1: all warnings being treated as errors
make[2]: *** [ideviceinstaller-ideviceinstaller.o] Error 1
make[2]: Leaving directory `/home/administrator/Desktop/libimobile_latest/ideviceinstaller-1.1.0/src'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/home/administrator/Desktop/libimobile_latest/ideviceinstaller-1.1.0'
make: *** [all] Error 2
administrator@1F3WXX1:~/Desktop/libimobile_latest/ideviceinstaller-1.1.0$ make  all-recursive
Making all in src
[... 26 lines cut ...]
   ^
In file included from ideviceinstaller.c:45:0:
/usr/local/include/libimobiledevice/installation_proxy.h:325:19: note: expected ‘instproxy_status_cb_t’ but argument is of type ‘void (*)(const char *, void *, void *)’
 instproxy_error_t instproxy_archive(instproxy_client_t client, const char *appid, plist_t client_options, instproxy_status_cb_t status_cb, void *user_data);
                   ^
ideviceinstaller.c:1391:3: error: passing argument 4 of ‘instproxy_restore’ from incompatible pointer type [-Werror]
   instproxy_restore(ipc, appid, NULL, status_cb, NULL);
   ^
In file included from ideviceinstaller.c:45:0:
/usr/local/include/libimobiledevice/installation_proxy.h:349:19: note: expected ‘instproxy_status_cb_t’ but argument is of type ‘void (*)(const char *, void *, void *)’
[... cut ...]
/usr/local/include/libimobiledevice/installation_proxy.h:372:19: note: expected ‘instproxy_status_cb_t’ but argument is of type ‘void (*)(const char *, void *, void *)’
 instproxy_error_t instproxy_remove_archive(instproxy_client_t client, const char *appid, plist_t client_options, instproxy_status_cb_t status_cb, void *user_data);
                   ^
cc1: all warnings being treated as errors
make[1]: *** [ideviceinstaller-ideviceinstaller.o] Error 1
make[1]: Leaving directory `/home/administrator/Desktop/libimobile_latest/ideviceinstaller-1.1.0/src'
make: *** [all-recursive] Error 1

can you please help me in this?

mexmer, 2016-07-22:
that error looks like if you compiling against different version of libimobile, i would suggest to not compile package, but use precompiled, they should work with ios10, if i'm looking correctly at version

http://packages.ubuntu.com/search?keywords=libimobile&searchon=names&suite=trusty&section=all

harjotcheema, 2016-08-12:
I downloaded and installed the latest version of libimobiledevice but am getting the error "Could not connect to lockdownd. Exiting."

The command that I ran is:
ideviceinstaller -u <UDID of Device> -l

My environment is as below:
Mac Mini running OS X El Capitan (Version 10.11.6)
iPhone 5s running iOS 10 beta
Xcode 8 beta

Can you please help to resolve this issue?

zgoda-mobica, 2016-08-19:
Another problem with iOS 10 Beta 5, `ideviceinfo` fails with

```
GnuTLS error: Error in the pull function.
ERROR: Could not connect to lockdownd, error code -5
```

Debian Jessie armhf, gnutls 3.5.2-2

Works for iOS 9.3.x

zgoda-mobica, 2016-08-19:
Same result when built against openssl 1.0.2h-1 on Debian Jessie armhf.

zgoda-mobica, 2016-08-22:
Another kind of error I'm getting for idevicebackup2:

```
Backup directory is "/tmp/backup/ios"
ERROR: Could not connect to lockdownd, error code -256
```

Commit https://github.com/libimobiledevice/libimobiledevice/commit/00f8e5733f716da8032606566eac7a9e2e49514d, configured with `./autogen.sh --without-cython CFLAGS="-static -static-libgcc -static-libstdc++"`, Debian Jessie with build dependencies from testing.

zgoda-mobica, 2016-08-30:
No matter how do I configure build, I'm getting `GnuTLS error: Error in the pull function.` error, and this is on ARMv7 Debian Jessie. Disabling OpenSSL changes notning, `ideviceinfo` always fails with errcode -5.

pmnordkvist, 2016-09-05:
I have the same issue but now on iOS 9 as well. I thought it was the latest iOS patch 9.3.5 but I also get it on 9.3.2. 

Everything compiles fine but i can not use ifuse or ideviceinfo. I am considering downgrading GnuTLS to < 3.5 to see if that helps. Any other suggestions?

pmnordkvist, 2016-09-05:
OK. I finally started over on a clean jessie install and compiled from scratch. Of course everything work. Still does not know why I never got this to work on my previously install. This issue should remain closed.

sunflash, 2016-09-08:
I get the same error with macOS 10.12 sierra and device running iOS 10, and both macOS 10.12 and iOS 10 is not beta version, but final release GM version. I did try to reinstall through homebrew, packages is recompile with Xcode 8 final, same issues still exist.

shusso, 2016-09-13:
For me recompiling libimobiledevice for ios10.0.1 + xcode8 GM (in osx 10.11.5) cured the lockdown issue.

```
[libimobiledevice]$ xcode-select -p
/Applications/Xcode8GM.app/Contents/Developer

[libimobiledevice]$ ./autogen.sh
[libimobiledevice]$ make
/Applications/Xcode8GM.app/Contents/Developer/usr/bin/make  all-recursive
Making all in common
  CC       socket.lo
  CC       thread.lo
  CC       debug.lo
In file included from debug.c:36:
../src/idevice.h:30:10: fatal error: 'openssl/ssl.h' file not found
#include <openssl/ssl.h>
         ^
1 error generated.
make[2]: *** [debug.lo] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```
To fix the SSL issue 
```
[libimobiledevice]$ CFLAGS=-I/usr/local/opt/openssl/include LDFLAGS=-L/usr/local/opt/openssl/lib ./configure
[libimobiledevice]$ make
```

qqtester, 2016-10-21:
anyone could give a solution to this error "-5"

nikias, 2016-10-21:
@qqtester the solution is in this thread. Recompile libimobiledevice using latest code from git

qqtester, 2016-10-25:
@nikias, thx. it is working now.

cajus, 2016-10-26:
I've recompiled libimobiledevice on Fedora 24 using `configure --disable-static --disable-openssl --enable-dev-tools --without-cython`. Revision 2c16751. Running *ideviceinfo* results in:

```
$ ideviceinfo -d
GnuTLS error: Error in the pull function.
ERROR: Could not connect to lockdownd, error code -5
```

gnutls-3.4.16. Any idea what's wrong?

qqtester, 2016-10-27:
@cajus, I used openssl. Maybe gnutls has the same problem.
I manually delete all the lib of old version openssl. as which in /usr/local/lib/. Then reinstall the new version openssl. Rebuild and install the latest libimobiledevice. Finally,  ideviceinfo worked.

gitsop01, 2016-10-27:
@cajus, 
It does work on Fedora 24 using openssl version 1.02h as a dependence  when compiled with
standard configure settings.

On Fedora 24, If you compile the latest git code with the standard configure settings,
If using  x86_64, When latest git compiled ENSURE you remove the standard /usr/lib64/libimobiledevice  library and links
Ensure to take notice of the symbolic links on the standard library so you can recreate same on updated library.

Make install  will install the new lastest git library in /usr/local/lib - just use sudo to copy to /usr/lib64 and fix the symbolic links to the new library.

TheElectronWill, 2016-10-27:
@qqtester @gitsop01 Instead of playing around with system librairies, it would be much better to create an RPM Package ([look at the wiki](https://fedoraproject.org/wiki/How_to_create_an_RPM_package)) based on the libimobiledevice's source code.

I've successfully done that for libimobiledevice, libimobiledevice-devel and libimobiledevice-utils (I haven't tested every utility yet but they should work). If someone is interested I could put the RPMs files and the files used to create them on github.

cajus, 2016-10-27:
@qqtester @gitsop01 looks like the library melagement isn't that good. I just changed the original spec file and removed the "--without-openssl" and it works now. I tried that before, but the libimobiledevice source wasn't in sync with HEAD, so that didn't work either. So: thanks - it's working again ;-)

@TheElectronWill sure. Always rpm/deb/whatever. It's my production system. But I'm wondering why upstream (Fedora) is building against gnutls when its more or less broken.

gitsop01, 2016-10-27:
@TheElectronWill @cajus 

Yes it should be done by updating the rpm spec file but in this case 
The problem is that the existing Fedora rpm spec file for 1.20 needs updating. As it contains the 
--without-openssl. It will be updated when a new version of libimobiledevice is released, then a 
bugzilla report can be added for this, with patch etc

The problem at that the issues should be marked in groups with the group as the first part of the
Subject field so that it is easy to see what issues are Linux, Xcode, Windows etc. Even use Labels
are better than nothing. 

This is reference to an old change in Fedora for CrypoPolicy
https://fedoraproject.org/wiki/Changes/CryptoPolicy

Several other changes are coming in Fedora 25 and Fedora 26 change-sets.
https://fedoraproject.org/wiki/Changes/NSSCryptoPolicies
https://fedoraproject.org/wiki/Releases/26/ChangeSet#OpenSSL_1.1.0.

gitsop01, 2016-10-27:
It does work on Fedora 24 using openssl version 1.02h as a dependence
 when compiled with
standard configure settings.

On Fedora 24, If you compile the latest git code with the standard
configure settings,
If using  x86_64, When latest git compiled ENSURE you remove the
standard /usr/lib64/libimobiledevice  library and links
Ensure to take notice of the smybolic links 
Make install  will install the new lastest git library in
/usr/local/lib - just use sudo to copy to /usr/lib64 and fix the
symbolic links
to the new library. 

> > > I've recompiled libimobiledevice on Fedora 24 using configure --
disable-static --disable-openssl --enable-dev-tools --without-
cython. Revision 2c16751. Running ideviceinfo results in:
> 
> $ ideviceinfo -d
> GnuTLS error: Error in the pull function.
> ERROR: Could not connect to lockdownd, error code -5
> 
> 
> gnutls-3.4.16. Any idea what's wrong?
> 
> 
> 
>   
>   
55493e4bb","name":"GitHub"},"entity":{"external_key":"github/libimob
iledevice/libimobiledevice","title":"libimobiledevice/libimobiledevi
ce","subtitle":"GitHub repository","main_image_url":"https://cloud.g
ithubusercontent.com/assets/143418/17495839/a5054eac-5d88-11e6-95fc-
7290892c7bb5.png","avatar_image_url":"https://cloud.githubuserconten
t.com/assets/143418/15842166/7c72db34-2c0b-11e6-9aed-
b52498112777.png","action":{"name":"Open in
GitHub","url":"https://github.com/libimobiledevice/libimobiledevice"
}},"updates":{"snippets":[{"icon":"PERSON","message":"@cajus in
#346: I've recompiled libimobiledevice on Fedora 24 using `configure
--disable-static --disable-openssl --enable-dev-tools --without-
cython`. Revision 2c16751. Running *ideviceinfo* results
in:\r\n\r\n```\r\n$ ideviceinfo -d\r\nGnuTLS error: Error in the
pull function.\r\nERROR: Could not connect to lockdownd, error code
-5\r\n```\r\n\r\ngnutls-3.4.16. Any idea what's
wrong?"}],"action":{"name":"View
Issue","url":"https://github.com/libimobiledevice/libimobiledevice/i
ssues/346#issuecomment-256314617"}}}

TheElectronWill, 2016-10-27:
For all Fedora users: I've just done a little script to easily build RPM packages of libimobiledevice. You can [find it here.](https://github.com/TheElectronWill/libimobiledevice-rpm). It works on x86_64 and i686 (haven't tested i686).

samrocketman, 2016-11-01:
The root cause for me (on Ubuntu) was that libimobiledevice was built to use GnuTLS.  I tested even the HEAD development version of GnuTLS and it still does not work with iOS 10.  The solution is to rebuild using OpenSSL instead.

I documented my solution:
https://github.com/libimobiledevice/ifuse/issues/32

I filed a bug report on Launchpad for Ubuntu:
https://bugs.launchpad.net/ubuntu/+source/libimobiledevice/+bug/1638177

nikias, 2016-11-04:
The pairing record generation with newer GnuTLS was broken. I just fixed it with commit https://github.com/libimobiledevice/libimobiledevice/commit/23069d10341ce637fdad7321d447c53752dba48c.
Make sure to remove invalid paring records by deleting the appropriate files or call ```idevicepair unpair``` for the affected devices.

WolfangAukang, 2016-12-08:
@TheElectronWill solution for Fedora worked for me. The problem @zgoda-mobica mentioned had been solved thanks to the newest version (1.2.1), compared to the one on Fedora repo (1.2.0).