libimobiledevice/libimobiledevice #347: Unable to synchronisze the iPhone on Debian/Ubuntu

alexarnaud, 2016-07-20:
Dear all,

I'm a blind users of the iPhone 5S with iOS 9.3 and I'm trying to synchronize musics, videos and photos on my GNU/Linux box.
The iPhone is correctly detected but I cannot open it with caja.
If I take only photo I can access to the phone storage (photo). The phone mounted with the gphoto protocol only, so it is not detected by Banshee or Rhythmbox. FYI, I have no music on the device.

How can I transfer musics, photos and videos? Do I have missed something ?

Environment : 
- Debian Sid unstable / Ubuntu 16.04
- libimobiledevice 1.2.0 and tested from master

I obtain this error message in syslog, I don't know if it's important or not : 
"usbmuxd[4552]: [01:12:21.008][1] config_get_device_record: failed to read '/var/lib/lockdown/3907505e067475e769d903af0848e8f220755685.plist': No such file or directory"

Best regards.

mexmer, 2016-07-20:
you need to pair device first to create pair record (which is stored in aforementioned .plist file after succesfull pairing)

alexarnaud, 2016-07-20:
Le 20/07/2016 à 15:56, Josef Micka a écrit :
>
> you need to pair device first to create pair record (which is stored 
> in aforementioned .plist file after succesfull pairing)
>
So if I understand correctly, if I pair the device with a Windows 
machine, after that I will be able to sync with Rhytmbox or shotwell in 
bidirectional sync (from phone to computer or computer to phone) ?

Best regards.

-- 
Alex ARNAUD

mexmer, 2016-07-20:
no you need to use idevicepair to pair it on ubuntu, then you can try sync, idk how rhythmbox works, and idk what you trying to sync.

nikias, 2016-07-20:
Music sync doesn't work anymore. Apple changed the media sync process alot along time ago and there is currently no working implementation available.

alexarnaud, 2017-06-08:
@nikias : do you know if the new protocol is documented somewhere ?

Best regards.

nikias, 2017-06-09:
@alexarnaud unfortunately not, it also involves a proprietary hashing/crypto operation that is required to actually make the device accept the commands.

alexarnaud, 2017-06-09:
Le 09/06/2017 à 14:28, Nikias Bassen a écrit :
>
> @alexarnaud <https://github.com/alexarnaud> unfortunately not, it also 
> involves a proprietary hashing/crypto operation that is required to 
> actually make the device accept the commands.
>
So If I understand well there are no solutions to figure out this issue 
expect to try a reverse engineering ? What is your solution to 
synchronize your music from an iPhone from GNU/Linux ?

Best regards.

-- 
Alex ARNAUD
Visual-Impairment Project Manager
Hypra - "Humanizing technology"

alexarnaud, 2019-10-02:
@nikias Do you know if there are progresses on this since 2017?

Best regards.

cyrialize, 2020-04-18:
Is it possible to sync music? This article implies that there is: https://www.groovypost.com/howto/howto/sync-your-iphone-or-ipod-touch-in-ubuntu/. 

I've gotten idevice pair to pair with my iphone and ideviceinfo shows my iphone. Unfortunately, I have not been able to get it to appear on Rhythmbox.

Filbuntu, 2022-10-20:
Probably not:
https://libimobiledevice.org/#faq
Or did anyone find a solution?

nikias, 2022-10-20:
No, this is only working with really old iOS versions.

Filbuntu, 2022-10-20:
Thank you for the quick reply!
It is a pity it does not work 😒, hopefully there will be again a workaround in the future!

nikias, 2022-10-20:
I haven't looked into it recently nor in the past 5 years as you can tell from the above comments. It is possible, but there are some technical hurdles involved that require some reversing around that proprietary hashing, and I have not enough spare time to look into it...

Filbuntu, 2022-10-24:
I understand that your time is limited. Maybe somebody else has the skills and likes a challenge 😊.

wolfteeth, 2025-02-01:
A couple years later, and I find myself wondering what it would take to make some progress on this particular problem. I see two other tickets asking basically the same question (#735, #331).

If there is some documentation of what needs to be done, maybe folks in the community would step up to help and remove some of the burden. Data collection, analysis, etc., or even the actual development, once the problem is better understood. And perhaps folks would donate money toward the effort (I would). Even as people are moving more and more toward streaming subscriptions, there are a few of us dinosaurs who maintain a personal music collection. :)

mexmer, 2025-02-03:
@wolfteeth if you are reffering to media sync (like itunes desktop application does), anyone can do it. libimobile provides all api needed for communication with phone, as for sync itself, one need to understand format of itunes database, and how adding/removing entries is done, this functionality doesn't need need to be integrated into libimobile and nobody needs to wait for libimobile developers to do it.

there are already tools like imazing, easeus mobimover, xilisoft iphone magic and others that can do it.

wolfteeth, 2025-02-04:
@mexmer Notably all of those tools are proprietary, closed source, and for Windows and Mac only. There are no working solutions for Linux, just a lot of kludgy workarounds. So Linux users do seem to be waiting for _someone_ to do it.

mexmer, 2025-02-04:
yes i have mentioned them just to point out it's doable, and i 'm aware for which platforms they are available, but since there is no opensource solution for long time, you can guess it's not easy to do it.