libimobiledevice/libimobiledevice #349: Possible to manually upload photos to iPhone photo library?

pmnordkvist, 2016-07-25:
Hi guys and really great work with this project so far. I use it regularly as part of a backup product and it works just fine. 

One thing I have been trying to play around with is to manually upload photos to the DCIM folder when mounted. The files are uploaded as expected but do not show up anywhere. As expected as well I guess. I guess there is some need for a file index or something to kick off. Has anyone played around with this? Is it at all possible?

Naville, 2016-08-17:
yep there is file index. No idea how to alter the index though.

mexmer, 2016-08-17:
you might delete it and restart device.
note it will take some time, until index is rebuilt after device restart.

pmnordkvist, 2016-08-26:
OK. Good to know. Will play around a bit with it. Where is the file index located?

pmnordkvist, 2016-08-26:
It looks like that iOS in general is using Sqlite3 so I would assume that the index is based on that.

mexmer, 2016-08-26:
i believe it was in afc://Photos or afc://Photodata

and yes, it is sqlite since ios 6

pmnordkvist, 2016-08-26:
Thanks! Will get back to you with my results. 

I noticed that they also use Sqlite3 with Pragma WAL which is very efficient when working with fiel databases.