libimobiledevice/libimobiledevice #351: Beta MobileDevice.dmg breaks libimobiledevice access

cmublitz, 2016-08-01:
I tried using the Apple Configurator 2.3 beta 2 and it requires installing a beta version of MobileDevice.  After installation, lockdownd_client_new_with_handshake fails with the error -19 LOCKDOWN_E_INVALID_HOST_ID.  The /var/db/lockdown/ pairing plist records seems to have changed with the addition of a new WiFiMACAddress field.  I'm not sure what else has changed about it.

I'm running:
OSX 10.11.6
libimobiledevice 1.2.0
libplist 1.12
libusbmuxd-1.0.10
libxml2-2.9.2
libzip-0.11.2

nikias, 2016-08-01:
You need a fix in libimobiledevice that is only available when you compile it yourself. The 1.2.0 release is too old for that.

cmublitz, 2016-08-02:
Awesome!  Thank you for the info and your work on libimobiledevice.  Do you have a guess as to when the next full cut will be released?  I assume after iOS 10 officially drops?

cmublitz, 2016-12-01:
Sorry for the hassle, but do you know which commit makes this fix?  I'm hesitant to jump to the master, but I would be willing to edit a couple of my 1.2.0 source files to make this work.  I'd like to update to Sierra, but my program that relies on libimobiledevice 1.2.0 would break.

Alternatively, if a new cut of libimobiledevice is expected soon, I don't mind waiting a bit longer for it.

Thanks!