libimobiledevice/libimobiledevice #353: Is there a way to know the estimated size of the backup?

oscarjiv91, 2016-08-03:
I was looking into `ideviceinfo` but I can't find anything related to the backup size, or at least the number of files that will be copied. I need to add a progress bar to my app when the backup is in progress.

mexmer, 2016-08-03:
apple mobile backup does report progress back in percent

oscarjiv91, 2016-08-03:
How can I get that info? I only see in the logs the percentage of each file being copied

mexmer, 2016-08-03:
look at function mb2_set_overall_progress_from_message in 
https://github.com/libimobiledevice/libimobiledevice/blob/master/tools/idevicebackup2.c#L411

it's missing DLMessageCopyItem, where progress is 3rd value in array, but overall should give you idea, how it works.

note, if you increase loglevel during version exchange (set it to 1 or 2), it will even send names of stages (like preparing backup, copying files and such)

cjhnim, 2016-08-04:
There is another choice.
Consider to "DLMessageGetFreeDiskSpace" message.
If you report "0 size" response to DLMessageGetFreeDiskSpace, you will be able to extract backup size through error message from iOS.

mexmer, 2016-08-04:
@cjhnim while that will indeed produce result, it will cancel backup process, so you need to start it twice.

if op needs only total progress (not own progress based on data count), using what is already in there is better imo.

cjhnim, 2016-08-04:
@mexmer You're right. I misunderstood the question. Thank you for reply.

oscarjiv91, 2016-08-04:
Thank you very much! I will look into it. Thanks again!

nikias, 2016-09-23:
@oscarjiv91 idevicebackup2 will process the DLMessageGetFreeDiskSpace and reports back to the device if the disk has enough remaining capacity. If not, the device will abort the backup operation.