libimobiledevice/libimobiledevice #354: network support feature for libimobiledevice

ryotyankou, 2016-08-04:
Network Support WIP 9.3.3 Allow libimobiledevice to talk to a device over the network (aka iTunes WiFi Sync).

What does this mean, is this feature completely new to all iOS and status is work in progress, once release, it will support 9.3.3?

Or it is already support iOS version lower than 9.3.3, and 9.3.3 version Network support now status is WIP?

ryotyankou, 2016-08-06:
is this a stupid question and everyone think disdain to answer?

gwagener, 2017-01-11:
I believe this means the feature was added to iOS in version 9.3.3 and support for it in libmobiledevice is WIP.

shalinnijel1, 2017-10-24:
Will this let users enable "Sync Over WiFi" without having to go through iTunes?

shalinnijel1, 2017-10-24:
Super! Once completed hope you update it here as well. Thanks again. :)

yury, 2018-06-19:
@nikias any progress on that? Just in case, may be some help required?

dominik-ziegler, 2018-10-12:
Really looking forward to this feature too. Any news on that? Do you need any help?

ElvishJerricco, 2018-11-09:
@nikias Also interested in this.

guillaume-uH57J9, 2018-12-15:
It would be great to have network support. 
You can help if you have time by checking the contribute section on http://www.libimobiledevice.org/.
It mentions a few way to contribute : read the source code on github, submit patch / pull requests, join the IRC channel or mailing list.

ElvishJerricco, 2019-06-08:
Is there any kind of documentation on how the wifi protocol works? I wouldn’t mind doing some research on how to implement it, but I can’t find any info.

guillaume-uH57J9, 2019-07-28:
I suspect Apple uses a proprietary or undocumented protocol for that. 

If you have iTunes try enabling WiFi sync and inspect traffic to see port(s) are being used, this may give some hints. But without documentation from Apple supporting this would be either tedious or impossible. 

Personally I stopped hoping libimobiledevice would support WiFi sync or other features with recent iOS versions. My long term solution is wait for the iOS device to die, and buy a device from another brand that has a more open platform, not something vendor-lock like Apple sells.