libimobiledevice/libimobiledevice #356: Unable to run commands without using sudo: "ERROR: lockdownd, -21" or "Could not start service com.apple.syslog_relay."

nnorton-home, 2016-08-10:
For the last week, I have been getting these errors when trying to run most libimobiledevice commands:

```
QA01:~$ ideviceinfo 
ERROR: Could not connect to lockdownd, error code -21

QA01:~$ idevicesyslog
ERROR: Could not start service com.apple.syslog_relay.
Could not start logger for udid <device_UDID>
```

LOCKDOWN_E_INVALID_HOST_ID = -21, (http://www.libimobiledevice.org/docs/html/lockdown_8h.html)

- Commands worked fine before last week, but I am not sure what I did to cause this.
- I've checked to make sure that "/usr/local/Cellar/" dir, and all enclosed dirs, have read & write permissions.
- I've uninstalled and reinstalled all packages and dependencies, and issue persists.
- This issue occurs for all iOS devices (and does not occur on other machines using the same devices)
- All commands work when run as "sudo", but that is not a viable solution for me.

**Packages (brewed)**:
libimobiledevice (1.2.0)
libplist (1.12)
libtool (2.4.6)
libusb (1.0.20)
automake (1.15)
autoconf (2.69)
openssl (1.0.2)
usbmuxd (1.0.10)
pkg-config (0.29)
gcc (6.1.0)

**System**:
mac OSX (10.11.6)

Does anyone have any idea what might be going wrong?

Gogle, 2016-08-11:
I have exactly the same problem. I have my iPhone upgraded to iOS 9.3.4. I also have my Mac restarted. Then this problem shows up. But I believe it's not the 9.3.4 causing this problem, as libimobiledevice works fine with other iPhones with iOS 9.3.4 on other Macs.

ashwither, 2016-08-11:
I don't believe it is iOS version specific either. I have seen this issue on every device I've tried (iOS 8.3-9.3.4).

johnboiles, 2016-08-17:
Doesn't seem to be iOS version specific. Same error on iOS 10 beta 6.

This issue seems similar on Ubuntu. For them it was a `usbmuxd` version issue
https://github.com/libimobiledevice/libimobiledevice/issues/243

johnboiles, 2016-08-17:
I'm unable to get this to work even with `sudo`. Though with a different error code for `ideviceinfo`

```
 > sudo ideviceinfo
Password:
ERROR: Could not connect to lockdownd, error code -5
```

@ashwither would it be possible to get the versions of your brewed packages from one of the working machine.

ashwither, 2016-08-18:
I listed all packages in "/usr/local/Cellar/" for both working and not working machines, and included all packages regardless of whether or not I thought they were related. I have updated and added some packages since my original post, in an effort to try and fix the issue.

 **Package** | **Not Working** | **Working** 
-------------- | ---------------- | -----------------
| ant | n/a | 1.9.7, HEAD |
| autoconf | 2.69 | 2.69 |
| automake | 1.15 | 1.15 |
| gcc | 6.1.0, HEAD |  n/a |
| gettext | 0.19.8.1 [keg-only] | n/a |
| glib | 2.48.1 |  n/a |
| gmp | 6.1.1 | n/a |
| ideviceinstaller | 1.1.0, HEAD | 1.1.0, HEAD |
| ifuse | 1.1.3, HEAD | n/a |
| ios-sim | 3.1.1, HEAD | n/a |
| isl | 0.17.1, HEAD | n/a |
| libffi | 3.0.13, HEAD [keg-only] |n/a |
| libgpg-error | n/a | 1.22 |
| libimobiledevice | 1.2.0, HEAD | 1.2.0, HEAD |
| libksba | n/a | 1.3.4 |
| libmpc | 1.0.3 | n/a |
| libplist | 1.12, HEAD |  1.12, HEAD |
| libtasn1 | 4.9 | 4.8 |
| libtool | 2.4.6 | 2.4.6 |
| libusb | 1.0.20, HEAD |  1.0.20, HEAD |
| libxml2 | 2.9.4, HEAD [keg-only] | 2.9.4, HEAD [keg-only] |
| linyaml | n/a |  0.1.6 |
| libzip | 1.1.2 | 1.1.2 |
| makedepend | 1.0.5 | n/a |
| maven | n/a |  3.3.9 |
| mpfr | 3.1.4 |  n/a |
| node | 6.4.0, HEAD |  6.2.0, HEAD |
| openssl  | 1.0.2h [keg-only] | 1.0.2h [keg-only] |
| pcre  | 8.39, HEAD |  n/a |
| pkg-config | 0.29.1 | 0.29.1 |
| readline | 6.3.8 [keg-only] |  6.3.8 [keg-only] |
| rlwrap | 0.42 |  n/a |
| usbmuxd | 1.0.10, HEAD | 1.0.10, HEAD |
| wget |  1.18, HEAD | 1.17.1, HEAD |
| xz | 5.2.2 | n/a |

urmyfaith, 2016-08-22:
I Got the same error as @ashwither 

my device is iPhone5s and iOS9.3.4

```
➜  ~ ideviceinfo  
ERROR: Could not connect to lockdownd, error code -21
```

```
➜  ~ idevicesyslog 
ERROR: Could not start service com.apple.syslog_relay.
```

ashwither, 2016-08-22:
This issue persists for other users on the same machine, so is not tied to a specific user profile.

johnboiles, 2016-08-23:
I have the Xcode beta installed alongside normal Xcode. Not sure if that could be causing issues. Do y'all have the beta installed?

urmyfaith, 2016-08-24:
@johnboiles 
I have Xcode7.3 and Xcode8 beta6 at the same time.

This could be the reason?

hbhorse, 2016-08-24:
ashwither, 
When I upgrade my osx to 10.11.16, I can't mount my iphone to mac. Do you know what is the problem?

Gogle, 2016-08-24:
@johnboiles I also have both of them installed. I suspect this is the cause.

ashwither, 2016-08-24:
The machine with issues is running Xcode 6.2 (6C131e), 7.3 (7D175), and 8.0 Beta 6 (8S201h). I want to point out that the issue also occurred while using Xcode 8.0 Beta 5. Both working machines are only running Xcode 7.3 (7D175).

ashwither, 2016-08-24:
@hbhorse - I have not run into that problem myself. What iOS and Xcode version are you using?

ashwither, 2016-08-24:
I have moved Xcode 8.0 Beta 6 to the trash, emptied the trash, and restarted my machine, but the issue still persists. I wasn't sure if this would solve the problem even if it is Xcode Beta, but that alone does not appear to do it.

Gogle, 2016-08-24:
Try installing Xcode 8.0 Beta on a working machine. See what will happen.

ashwither, 2016-08-24:
Yeah I can try that on a new machine.

johnboiles, 2016-08-24:
#346 looks like maybe the same / similar as this

ashwither, 2016-08-24:
I can confirm that installing Xcode 8.0 Beta 6, on a working machine with libimobiledevice, does cause this issue to occur.

joshbax, 2016-08-24:
I'm experiencing this issue also with **Xcode 8, beta 6.** I just started using libimobiledevice as a workaround for Xcode no longer supporting plugins in Xcode 8, so I apologize in advance if I'm not telling anyone something they don't already know.

That said,  I'm able to get part of the library working by avoiding the lockdownd handshake. **Using `ideviceinfo -s` is working for me**. I'm still going through the source to figure out how to make idevicesyslog to work as well. Hope this helps.

ashwither, 2016-08-24:
Not sure if this is related, but I also am seeing two different Xcode paths depending on whether I run the command with or without sudo. 

```
QA01:~$ sudo xcode-select -p
/Applications/Xcode7.3.1.app/Contents/Developer

QA01:~$ xcode-select -p
/Applications/Xcode6.2.app/Contents/Developer
```

ashwither, 2016-08-25:
Does anyone have any ideas what changed in Xcode 8 that might be responsible for this? 

It seems to me that simply removing all the Beta components does not fix the issue. Whatever the solution is probably needs to revert a change made by Xcode 8 Beta.

vincentanupkurisc, 2016-08-25:
Facing the same problem once I installed xcode beta. Anyone looking into this?

ashwither, 2016-08-25:
Tracing failing libimobiledevice commands using opensnoop, points the finger to /var/db/lockdown/, which turns out to have different permissions than those of working machines.

```
QA01:~$ sudo opensnoop -n ideviceinfo

  UID    PID COMM          FD PATH     
1123489958   1067 ideviceinfo    4 /var/db/lockdown
```

**This issue can _only temporarily_ be fixed either by using chmod or repair_packages to change the permissions back**:

```
QA01:~$ sudo chmod 777 /var/db/lockdown
```
**or** 
```
QA01:~$ sudo /usr/libexec/repair_packages --repair --standard-pkgs --volume /
Password:
	Permissions differ on "private/var/db/lockdown", should be drwxrwxrwx , they are drwx------ .
	Repaired "private/var/db/lockdown".
```

**As soon as the machine reboots, the permissions change back.**

Gogle, 2016-08-29:
Pymobiledevice also fails with Xcode 8 beta. But it gives a different error msg:
`M2Crypto.RSA.RSAError: unknown message digest algorithm`

ssynix, 2016-09-01:
Also on **Xcode 8, beta 6** and facing the same problem. Using `sudo` or `chmod` has not fixed the issue for me.

joshbax, 2016-09-01:
@ssynix I believe you have to clone the source and manually build the latest version, which is 1.2.1. That is, if you can figure out how to get around the OpenSSL linker errors.  Not sure why this version hasn't been formally released yet.

ssynix, 2016-09-01:
I tried doing that this morning on my mac, but I ran into a lot of missing libraries and what appeared to be circular dependencies. I'll try it out later when I have more time.

Is there an easier way to do the build from source?

joshbax, 2016-09-01:
@ssynix I ran into a lot of the same issues. I eventually installed all the dependencies but the one I couldn't resolve was a missing`openssl/ssl.h`. Then when I searched that issue, I fell into wormhole of reading [stuff like this](https://langui.sh/2015/07/24/osx-clang-include-lib-search-paths/) and gave up.

ssynix, 2016-09-07:
I managed to get it working by:
```shell
brew install autoconf automake libtool pkg-config
# to make the openssl link error go away:
CFLAGS=-I/usr/local/opt/openssl/include LDFLAGS=-L/usr/local/opt/openssl/lib ./configure
```

cbracken, 2016-09-21:
The underlying issue appears to be with the version of libimobiledevice available via homebrew. The simplest workaround until a new release is cut is likely to install from head:
```
brew uninstall libimobiledevice
brew install --HEAD libimobiledevice
```

I expect getting a 1.2.1 release tarball up would be enough to get an updated [formula](https://github.com/Homebrew/homebrew-core/blob/master/Formula/ideviceinstaller.rb) out. #380 looks like it tracks that.

qqtester, 2016-09-21:
@joshbax , maybe you need copy the folder of openssl which include .h files into project folder. I just fix this. The path is /usr/local/Cellar/openssl/1.0.2h_1/include/

joshbax, 2016-10-03:
@qqtester I was able to get it working by using the exact CFLAGS/LDFLAGS mentioned earlier. I had tried this previously and it didn't work, probably a typo. Thanks.

franklinyu, 2017-11-01:
Duplicate of #188