libimobiledevice/libimobiledevice #358: Unable to build with gnutls

zgoda-mobica, 2016-08-17:
If I try to build with gnutls (--disable-openssl) on Debian 8.0 (Jessie), configure fails with:

```
checking gcrypt.h usability... yes
checking gcrypt.h presence... yes
checking for gcrypt.h... yes
[ ...skipped... ]
checking for openssl... yes
*** Note: OpenSSL support explicitly disabled ***
checking for libgnutls... yes
checking for gcry_control in -lgcrypt... no
configure: error: libgcrypt is required to build libimobiledevice with GnuTLS
```

libgcrypt20 and libgcrypt20-dev are both installed. How do I build against GnuTLS?

zgoda-mobica, 2016-08-19:
The same works for Ubuntu 16.04 amd64. The procedure is basically:

```
$ apt-get build-dep libimobiledevice-utils
$ ./autogen.sh --without-cython --disable-openssl
```

libgcrypt20 and libgcrypt20-dev are older (1.6.5-2)

Looks like libgcrypt is somehow broken on Debian armhf, can someone confirm before I report this in debian-gnutls?

zgoda-mobica, 2016-08-19:
Okay, found it. It seems related to static building. It succeeds if I build dynamically linked binary but fails for "partially static" version (with CFLAGS="-static -static-libgcc -static-libstdc++").

closed 2016-09-27 by nikias

fikretsengul, 2017-04-20:
Hey, I had some problems when was trying to compile libimobiledevice with openssl (linking problems.) For exp.

CCLD libimobiledevice.la
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libimobiledevice.la] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

I found a solution, which disables openssl. For exp.

brew install gnutls 
brew install libgcrypt
./configure --disable-openssl

But when I enter the third code, it gives me the same error that you had. For exp.

checking for gcry_control in -lgcrypt... no
configure: error: libgcrypt is required to build libimobiledevice with GnuTLS

You solved that but how ? I didn't understand. Can you help me ?