libimobiledevice/libimobiledevice #359: Detect wrong SIM status on iPad Pro 9.7 inch (ipad6,4) iOS 9.3.4

truonggiang0710, 2016-08-19:
Hi guys,

I have an iPad Pro 9'7 **without SIM** (OS 9.3.4).
But I ran "**ideviceinfo -k SIMStatus**" to check SIM status. 
The result is "**kCTSIMSupportSIMStatusReady**". It looks like as phone has a SIM.
Please help me to verify SIM status exactly ! I'm confused !

Thanks a lot !

mexmer, 2016-08-19:
run ideviceinfo -d to see communication, but most likely device is reporting it wrong, since that status looks like picked from device answer.

truonggiang0710, 2016-08-19:
Thanks @mexmer but
I tested all iDevices. It's OK and exactly. Except iPad Pro 9'7.
I think iPad Pro 9'7 has an embedded Apple SIM. Maybe this reason make me read SIM status wrong.
I don't know any way to verify this.

mexmer, 2016-08-19:
like i said, run ideviceinfo -d
it will show you communication, and it will show you response from device.
look SIMStatus in response - whatever it says is from device, not from imobile library - so if value reported is wrong, then you can only ask apple to fix it.

this is snip from response on iPhone
```
<key>SDIOProductInfo</key>
<string>P=N90 m=3.1 V=m</string>
<key>SIMGID1</key>
<data>
/////////////w==
</data>
<key>SIMGID2</key>
<data>
/////////////w==
</data>
<key>SIMStatus</key>
<string>kCTSIMSupportSIMStatusReady</string>
<key>SoftwareBehavior</key>
<data>
AQAAAAAAAAAAAAAAAAAAAA==
</data>
<key>SoftwareBundleVersion</key>
<string></string>
<key>SupportedDeviceFamilies</key>
<array>
<integer>1</integer>
</array>
<key>TelephonyCapability</key>
<true/>
```

truonggiang0710, 2016-08-22:
Thanks @mexmer ! I found out other way to detect exactly SIM status in my case.

Thanks for your help again !

closed 2016-08-22 by truonggiang0710