libimobiledevice/libimobiledevice #360: Can libimobiledevice get the installed apps?

lianzisong, 2016-08-22:
what I want is scan the set to get the installed apps(including the installed ever).
Can libimobiledevice support it? or does the IOS sets support it?

dweinstein, 2016-08-22:
the currently installed apps list you can get from `ideviceinstaller` utility -- see `ideviceinstaller -l`. it won't tell you previously installed apps though.

lianzisong, 2016-08-22:
@dweinstein thanks, so it's impossible to get the previously installed apps through the interface.
can I get the tips from by checking the folders name ? like scanning the directories name compare the apps package name?  is it possible?

lianzisong, 2016-08-22:
@dweinstein BTW ,I found that the backup will generate the Manifest.plist ,this file seems record the apps of the set ever installed , cause I found the package name I ever installed then uninstalled in this file.
Can the Manifest.plist tell that which apps were uninstalled ?

Kanong017, 2016-08-29:
中国人..加群450479816一起讨论