libimobiledevice/libimobiledevice #363: Does libimobiledevice has plan to support "com.apple.dt.simulatelocation" service?

nghiadhd, 2016-08-26:
I'm going to implement a location spoof look like "Simulate Location" of Xcode for "Automation Testing" purpose

I found that Xcode connect to "com.apple.dt.simulatelocation" service to simulate location.
I wrote a sample base on libimobiledevice and I can connect to "com.apple.dt.simulatelocation" service. However I don't know the protocol to communicate with "com.apple.dt.simulatelocation". I referred some other services but seem each of them has specific protocol

My question that how do you know the protocol of "screenshot/debugserver/sbs/..."? If possible, please share with me

Kanong017, 2016-08-29:
Need to compile a new version to support ios10

testobjectnishant, 2016-12-10:
@nghiadhd hey, I am working on the same solution. Did you able to make any progress? 
I also want to know the messages that needs to be send to inject the location.

qmfrederik, 2017-03-29:
I think #480 fixes this