libimobiledevice/libimobiledevice #365: Can not connect to lockdownd

jmukes97, 2016-08-28:
I am on Kali Linux 2.0 My iphone is on the most recent beta for ios 10. When i try to backup I get the following error. 

GnuTLS error: Error in the pull function.
ERROR: Could not connect to lockdownd, error code -5

What am i doing wrong? I installed libimobiledevice  from http://www.libimobiledevice.org/ and installed the dependencies with terminal. If anymore information is needed ill be sure to share.

Kanong017, 2016-08-29:
Need to compile a new version to support ios10

zgoda-mobica, 2016-08-29:
I get the same error with current repo HEAD.

jmukes97, 2016-08-29:
@Kanong017 Where can i find the new version to compile?

VasilijSviridov, 2016-08-31:
please provide steps how to compile new for iOS10 on mac osx 

Cheers

joecool1029, 2016-09-19:
I hit same issue and it took two evenings to figure out. I think nettle actually screwed up. I moved gcc versions since I compiled nettle last. Now I just rebuilt the full stack below with clang. 

Rebuild nettle, rebuild gnutls. Then rebuild all the libimobiledevice stuff and you should be good to go.

VasilijSviridov, 2016-09-19:
ok, this worked for me, please check https://github.com/appium/appium/issues/6597#issuecomment-247038942

:)

Justin-Maxwell, 2017-10-05:
Possible workaround - I had a similar (I think identical) error.  Forcing OpenSSL via build using `./autogen.sh --enable-openssl` seemed to resolve it.

Stumbling across this comment lead me to try it 
https://github.com/libimobiledevice/ifuse/issues/32#issuecomment-258393182