libimobiledevice/libimobiledevice #367: [lockdown] Error: saving pair record failed: -22

MarcusXie, 2016-09-02:
I'm trying to port libimobiledevice and usbmuxd to a specific android platform.
a few problems happens.

Now when usbmuxd detected the Apple device, 
1. And trying perform [idevicepair pair] failed.
2. and error log:  **D/libusbmuxd(  234): usbmuxd_save_pair_record: Error: saving pair record failed: -22**
3. then print [SUCCESS: Paired with device c1ab1f256b7c63f3c38e2feae2cd3dca377ca308] .
4. and no 40bit.plist generated.

any hint? thanks.

mexmer, 2016-09-02:
saving pair record failed means, that usbmuxd doesn't have write access to directory, where you define it should save pairing data,

it can be either permission issue, or simply that directory doesn't exists.

MarcusXie, 2016-09-09:
I modified as Ubuntu do, and problem solved.
Issue close.

closed 2016-09-09 by MarcusXie

joeasy, 2016-12-16:
I have same problem on android platform, could you please kindly tell me how to solve this problem "usbmuxd_save_pair_record: Error: saving pair record failed: -22".

MarcusXie, 2016-12-23:
I do it as ubuntu do, which is different from android(I think).
You could run "apt-get source usbmuxd" such thing, to get the source code from ubuntu.
apt-get source libimobiledevice and libusbmuxd2 maybe.