libimobiledevice/libimobiledevice #368: Failed to work on iPad Pro (9.7-inch) (Model A1673)

ishahak, 2016-09-04:
Hi,
Although pairing works ok:
```
idevicepair -d pair
SUCCESS: Paired with device 8c19996cd91729f454a5d7f411d1b5367b729d51

```
and ideviceinfo without pairing is ok:
```
ideviceinfo -s
BoardId: 8
BuildVersion: 14A5346a
ChipID: 32769
DeviceClass: iPad
DeviceColor: #272728
DeviceName: Ehud's iPad Pro
DieID: 7903599489458726
HardwareModel: J127AP
PartitionType: 
ProductName: iPhone OS
ProductType: iPad6,3
ProductVersion: 10.0
ProductionSOC: true
ProtocolVersion: 2
TelephonyCapability: false
UniqueChipID: 7903599489458726
UniqueDeviceID: 8c19996cd91729f454a5d7f411d1b5367b729d51
WiFiAddress: 98:01:a7:0a:69:ed

```
All other tools are failing:
```
ideviceinfo 
ERROR: Could not connect to lockdownd, error code -21

```
and
```
idevicesyslog 
ERROR: Could not start service com.apple.syslog_relay.
Could not start logger for uuid 8c19996cd91729f454a5d7f411d1b5367b729d51

```
Any idea please?
Thank you!

truonggiang0710, 2016-09-06:
I think you should be run with last version.
Download last version (libimobiledevice, libusbmuxd, usbmuxd,.. ) and run again !

ishahak, 2016-09-06:
I was using `brew install libimobiledevice`, and expecting it to bring the most up-to-date components. If this is not the case then indeed this issue can be closed.