libimobiledevice/libimobiledevice #369: How Can I upload the file into the filesystem?

lianzisong, 2016-09-07:
Hi ,guys.
cause the iTunes on the windows support the application file uploading , so I wonder whether the libimobiledevice support it? and how to do it?

PS:
upload the application file means:
In the iTunes environment ,drag the files on PC into the APP's upload area ,then ,click the sync button . the files will be uploaded into the IOS device.

Kanong017, 2016-09-08:
lockdownd_start_service 开启com.apple.mobile.house_arrest服务 得到句柄house
house_arrest_client_new 初始化服务和设备的连接并得到最终的house句柄(device,house,@house_handle)
house_arrest_send_command 发送VendDocuments命令
afc_client_new_from_house_arrest_client 将house_handle和afc服务连接

然后就是正常的上传文件,不需要同步...
必须要应用开启了UIFileSharingEnabled才能连接到他的Documents目录...
至于更详细的...
加群450479816

mexmer, 2016-09-08:
you can use ifuse to mount document sharing for particular application, since iOS 8.3 only applications that have document sharing enabled can be mounted this way, and you can upload data to them.

also note, that due changes in iOS encryption, you will need to have device screen unlocked during read/write on mounted application storage.

lianzisong, 2016-09-09:
@mexmer cause I want to embedded this function into my project , and prefer the process can be control by my code ,so the ifuse is not suitable for me.. thanks anyway .

lianzisong, 2016-09-09:
@Kanong017 谢谢，好像我已经加了。但是群里没什么人说话。
github上也有example，我找找看。
[https://github.com/aburgh/libimobiledevice/edit/master/tools/afccl.c](url)

mexmer, 2016-09-09:
then as @Kanong017 sugested, using housearrest service, which is what ifuse does.

you can check ifuse sourcecode
https://github.com/libimobiledevice/ifuse

daniel-bluesea, 2016-09-20:
@lianzisong, @mexmer is there a similar way to put waiting icon for applications restored to a new device?

mexmer, 2016-09-20:
@daniel-bluesea i'm not aware of such thing. there might be command in installation proxy tho'

closed 2016-10-31 by lianzisong