libimobiledevice/libimobiledevice #371: Modifying libimobiledevice to look for dependency libraries elsewhere

OliverXia, 2016-09-13:
Hello!

I've been trying to get libimobiledevice.dylib to look for its dependencies (e.g. Libgcrypt and others) elsewhere instead of /usr/local/. Does anyone know what I need to configure to do that? I am using libimobiledevice.dylib in a program I wrote, it's just an executable. And I package all the libraries that it needs with the executable during the "make" phase. I want my executable that's using libimobiledevice to look in my executable's directory for the dependency libraries as opposed to /usr/local/. That way, people can just use the executable without having to download or fiddle with installing the dependency libraries. Any help is appreciated!!

gitsop01, 2016-09-16:
A better approach would be to use flatpak or Snappy (package manager) this way the dependencies and run time environment are always correct and if done correctly in a sandbox should be fairly secure.
Downside is both are fairly new technologies with rapid changes etc but look promising.

flatpak.org

OliverXia, 2016-09-16:
Ah! The way I decided to do it was use install_name_tool to redirect the dependencies of the dylibs to look for those dependencies in the executable's directory. That was fairly painless, but install_name_tool took a while to figure out how to use.

closed 2016-09-16 by OliverXia