libimobiledevice/libimobiledevice #372: iOS10 on Linux, Trust record not preserved through device restart

progressnerd, 2016-09-13:
To reproduce:
1. Plug iOS10 device into a Linux machine with the latest usbmuxd and libimobiledevice installed
2. Press the "Trust" button on the device when the Trust dialog appears
3. Run `idevicedate` to demonstrate idevice commands work
4. Run `idevicediagnostics restart` to restart the device
5. Once the device has restarted, running `idevicedate` produces:
`ERROR: Could not connect to lockdownd, error code -19`

Repeating Step 5 with the "-d" debug flag displays output ending with:
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Error</key>
	<string>PairingDialogResponsePending</string>
	<key>Request</key>
	<string>Pair</string>
</dict>
</plist>
15:28:39 third_party/libimobiledevice/src/lockdown.c:188 lockdown_check_result(): ERROR: PairingDialogResponsePending
15:28:39 third_party/libimobiledevice/src/lockdown.c:986 lockdownd_do_pair(): Pair failure
15:28:39 third_party/libimobiledevice/src/lockdown.c:726 lockdownd_client_new_with_handshake(): Device shows the pairing dialog.
```

And upon pressing the Home button on the device, a new Trust dialog appears.

yzhao2012, 2016-10-25:
I'm also getting this problem. Is there anyway to remove the trust option on the device?

progressnerd, 2016-10-25:
No way to remove the trust option on the device. Not Apple's style :) It would be cool if usbmuxd offered a way to cache the trust in memory and replay it (perhaps also re-writing it to disk if it disappeared), to overcome flakiness in trust pairing and the spontaneous Trust dialogs we see, even on Macs.

jevinskie, 2016-10-25:
IIRC my company's automated testing started to see this with iOS 9 (maybe 8?) when using `idevicediagnostics restart` with macOS host.

yzhao2012, 2016-10-25:
I actually saw this a couple times with iOS9. For some reason it seems slightly finicky to me. When I end up running `idevicediagnostics restart ` from my macSierra host, I usually have to repair. But I just tested this on an El Capitan OS and everything seemed fine. Maybe it's iOS related?

IronLung7, 2016-11-17:
Having this problem with IOS 9.3.2 and windows 7

IronLung7, 2016-11-23:
@yzhao2012  Are you using the same version of libimobiledevice on these two OS X?

andrewllyons, 2017-01-04:
Can confirm this is also happening for me, I will investigate more into the source to try and find why this happens, for now I'm disabling restart in our automation scripts.

mexmer, 2017-01-11:
see following, regarding lockdown records

https://articles.forensicfocus.com/2016/11/14/forensic-implications-of-ios-lockdown-pairing-records/

TL;DR
if you have password protected (or fingeprint protected) iOS device, you need to unlock device first, prior trying to communicate with it. pairing records are not working on locked iOS devices after restart until you unlock them at least once.

ohadschn, 2017-02-02:
@mexmer this happens on devices that aren't protected (neither password nor fingerprint).

Is there some way to work around this? I have to restart the device before my automated tests due to https://github.com/appium/appium/issues/7192, but the loss of trust means manual intervention is necessary...

andrewllyons, 2017-02-07:
@ohadschn One work I've managed to perform consistently, albeit a hack, is to build a dummy app and install it on the device, which upon launching takes control of the active screen context and unlocks the phone and immediately terminates, effectively unlocking via this potential flaw in apples security. The process is to restart, monitor idevice_id for the UID, and once reconnected mount the developer image and launch the app. I don't like the method, because it can occasionally fail to mount a developer image due to seemingly randomly losing its trust privleges with AFC error 19. (No trust). I'll continue to look into alternative solutions from the libi code side.

ohadschn, 2017-02-08:
@andrewllyons thanks! Any chance you could post the source of that app and the script you're using to "monitor idevice_id for the UID, and once reconnected mount the developer image and launch the app"?