libimobiledevice/libimobiledevice #374: Can no longer see Documents after iOS 10 update

polyx, 2016-09-14:
When I would connect my iDevices running iOS 9 I would be able to access Documents with app folders. After updating iPhone 6s to iOS10 I can only access photos.
I have latest libimobiledevice installed.

mikch06, 2016-09-19:
Same problem. There appears a GNU Tls pull error. May a handshake exception.

robm92, 2016-09-20:
Apple have deprecated use of TLS from iOS10 and MacOS 10.12 which is likely the issue.

mexmer, 2016-09-20:
@robm92 that's rather doubltfull
since there is no other transport security than TLS used wordlwide, SSL being deprecated everywhere due large number of security flaws.

since version 10 iOS dropped support for SSLv3 and use TLS 1.0+

my app (based on libimobile) can access application sandboxes even on iOS 10, and yes, i can confirm it is using TLS 1.0 for secure communication.

polyx, 2016-09-20:
For some reason  it started working today, might be because of some recent updates for ubuntu, but I am not sure if they did any TLS related updates

robm92, 2016-09-21:
@mexmer My mistake, Apple dropped SSLv3, not TLS

rknall, 2016-09-23:
@polyx - Which version of Ubuntu are you using?

polyx, 2016-09-23:
@rknall 16.04

mexmer, 2016-09-23:
@polyx , check if either gnutls/openssl or libimobile was updated, one of them it should be.

some months ago in libimobile was sslv3_method(which does use sslv3 only) changed to sslv23_method(which does TLS/SSL negotiation, but breaks compatibility with elder ios device - i can confirm, it doesn't pair/work with ios 5, but works from ios7 onward - mind up to ios 9 apple used sslv3).

i'm lazy to check source, but i think, this change should be in latest release of libimobile, there are other changes related to this in head tho'

polyx, 2016-10-01:
@mexmer Sorry for not responding sooner. No there were no updates to packages you mentioned until 23sep. which was after everything started working (it was update to openssl). Perhaps after installing libimobiledevice from sources something needed restart and since I dont restart often it didn't work until than?

deepinzhangshuang, 2016-10-20:
I have update gvfs to 1.30.1.1-1 for debian,  gvfs can mount IOS 10 for gphoto.  the libimobilidevice update do nothing for mount .