libimobiledevice/libimobiledevice #376: idevicebackup2 failing with iOS 10 (Error code 104)

DaKaZ, 2016-09-15:
I am using the latest copy of libimobiledevice on OS X, compiled from source.  Since upgrading a device to iOS 10, I am unable to backup getting error code 104.  Yes, the original directory is empty (you'll see I create the directory first).  Any ideas on why?

```
$ mkdir /tmp/backups
$ ./idevicebackup2 -d backup --full /tmp/backups
Backup directory is "/tmp/backups"
Started "com.apple.mobilebackup2" service on port 51154.
Negotiated Protocol Version 2.1
Starting backup...
Enforcing full backup from device.
Backup will be unencrypted.
Requesting backup from device...
Full backup mode.
[===========================                       ]  54% Finished
Receiving files
[==================================================] 100% (9.1 MB/9.1 MB)       
[===================================               ]  68% Finished
Receiving files
[==================================================] 100% (35.8 KB/35.8 KB)     
[=====================================             ]  74% Finished
Receiving files
[==================================================] 100% (2.2 MB/2.2 MB)       
[==================================================] 100% (2.2 MB/2.2 MB)     
[==================================================] 100% (2.2 MB/2.2 MB)     
[==================================================] 100% (2.2 MB/2.2 MB)     
[==================================================] 100% (2.2 MB/2.2 MB)     
[==================================================] 100% (2.2 MB/2.2 MB)     
[==================================================] 100% (2.2 MB/2.2 MB)     
[=========================================         ]  82% Finished
Receiving files
[==================================================] 100% (4.5 MB/4.5 MB)        
[==================================================] 100% (4.5 MB/4.5 MB)     
[==================================================] 100% (4.5 MB/4.5 MB)     
[==================================================] 100% (4.5 MB/4.5 MB)     
[==================================================] 100% (4.5 MB/4.5 MB)     
[==============================================    ]  91% Finished
[==============================================    ]  91% Finished
[==============================================    ]  91% Finished
[==============================================    ]  91% Finished
Moving 128 files
[===============================================   ]  93% Finished
Moving 128 files
[================================================  ]  95% Finished
Moving 128 files
[================================================= ]  97% Finished
Moving 107 files
[==================================================]  99% Finished
Moving 1 file
[==================================================]  99% Finished
Moving 1 file
[==================================================]  99% Finished
[==================================================] 100% Finished
Removing 1 file
Could not remove '/tmp/backups/1e0c886418298692324f330b38b32196c89bd37f/Snapshot': Directory not empty (66)
[==================================================] 100% Finished
ErrorCode 104: Error removing snapshot directory (MBErrorDomain/104). Underlying error: Directory not empty (MBErrorDomain/100).
Received 496 files from device.
Backup Failed (Error Code 104).
```

looking at the directories we have this:
```
$ ls /tmp/backups/1e0c886418298692324f330b38b32196c89bd37f/
00		21		42		63		84		a0		c1		e2
01		22		43		64		85		a1		c2		e3
02		23		44		65		86		a2		c3		e4
03		24		45		66		87		a3		c4		e5
04		25		46		67		88		a4		c5		e6
05		26		47		68		89		a5		c6		e7
06		27		48		69		8a		a6		c7		e8
07		28		49		6a		8b		a7		c8		e9
08		29		4a		6b		8c		a8		c9		ea
09		2a		4b		6c		8d		a9		ca		eb
0a		2b		4c		6d		8e		aa		cb		ec
0b		2c		4d		6e		8f		ab		cc		ed
0c		2d		4e		6f		90		ac		cd		ee
0d		2e		4f		70		91		ad		ce		ef
0e		2f		50		71		92		ae		cf		f0
0f		30		51		72		93		af		d0		f1
10		31		52		73		94		b0		d1		f2
11		32		53		74		95		b1		d2		f3
12		33		54		75		96		b2		d3		f4
13		34		55		76		97		b3		d4		f5
14		35		56		77		98		b4		d5		f6
15		36		57		78		99		b5		d6		f7
16		37		58		79		9a		b6		d7		f8
17		38		59		7a		9b		b7		d8		f9
18		39		5a		7b		9c		b8		d9		fa
19		3a		5b		7c		9d		b9		da		fb
1a		3b		5c		7d		9e		ba		db		fc
1b		3c		5d		7e		9f		bb		dc		fd
1c		3d		5e		7f		Info.plist	bc		dd		fe
1d		3e		5f		80		Manifest.db	bd		de		ff
1e		3f		60		81		Manifest.plist	be		df
1f		40		61		82		Snapshot	bf		e0
20		41		62		83		Status.plist	c0		e1
```
and
```
$ ls /tmp/backups/1e0c886418298692324f330b38b32196c89bd37f/Snapshot/
00	10	20	30	40	50	60	70	80	90	a0	b0	c0	d0	e0	f0
01	11	21	31	41	51	61	71	81	91	a1	b1	c1	d1	e1	f1
02	12	22	32	42	52	62	72	82	92	a2	b2	c2	d2	e2	f2
03	13	23	33	43	53	63	73	83	93	a3	b3	c3	d3	e3	f3
04	14	24	34	44	54	64	74	84	94	a4	b4	c4	d4	e4	f4
05	15	25	35	45	55	65	75	85	95	a5	b5	c5	d5	e5	f5
06	16	26	36	46	56	66	76	86	96	a6	b6	c6	d6	e6	f6
07	17	27	37	47	57	67	77	87	97	a7	b7	c7	d7	e7	f7
08	18	28	38	48	58	68	78	88	98	a8	b8	c8	d8	e8	f8
09	19	29	39	49	59	69	79	89	99	a9	b9	c9	d9	e9	f9
0a	1a	2a	3a	4a	5a	6a	7a	8a	9a	aa	ba	ca	da	ea	fa
0b	1b	2b	3b	4b	5b	6b	7b	8b	9b	ab	bb	cb	db	eb	fb
0c	1c	2c	3c	4c	5c	6c	7c	8c	9c	ac	bc	cc	dc	ec	fc
0d	1d	2d	3d	4d	5d	6d	7d	8d	9d	ad	bd	cd	dd	ed	fd
0e	1e	2e	3e	4e	5e	6e	7e	8e	9e	ae	be	ce	de	ee	fe
0f	1f	2f	3f	4f	5f	6f	7f	8f	9f	af	bf	cf	df	ef	ff
```
Happy to do any code changes / troubleshooting if someone can point me in the correct direction...

DaKaZ, 2016-09-15:
I just noticed that this is addressed by PR: https://github.com/libimobiledevice/libimobiledevice/pull/362/commits/88a470225b14efd7df5c042991226fc0a88d8a20

nikias, 2016-09-18:
Fixed with commit https://github.com/libimobiledevice/libimobiledevice/commit/6aecc7228dbb30b4db48374ffb4ae4a1616bdd95

closed 2016-09-18 by nikias

DaKaZ, 2016-09-19:
Perfect!  I have tested and this works great.  Thanks @nikias !!

Erfanthe1, 2023-01-18:
i receive the same error during back up for my iphone 
can you help me fix it?