libimobiledevice/libimobiledevice #377: ERROR: Could not connect to lockdownd, error code -2

ROM-PacMe, 2016-09-19:
[image]

MidiGlitch, 2016-09-19:
Have installed Xcode 8 alongside 7.3 and am also getting lockdown error (code -21).
"ideviceinfo" does not work but ideviceinfo -s does work

lukcioj, 2016-09-21:
Solution for this issue is here: https://github.com/libimobiledevice/libimobiledevice/issues/356#issuecomment-248486535

```
brew uninstall libimobiledevice
brew install --HEAD libimobiledevice
```

qqtester, 2016-10-10:
@MidiGlitch , update openssl maybe fix this.

moofnbg, 2016-10-11:
hi, i have the same problem. Code -13 and -21. Can someone help? Cannot use appium to automate some web testing...:-(

moofnbg, 2016-10-11:
This solved the problem for me: 

_brew uninstall libimobiledevice
brew install --HEAD libimobiledevice_

songhaonangit, 2016-11-16:
ubuntu 14.04 with the latest code get lockdown error (code -21).

walkjivefly, 2017-09-17:
Getting the same error code -2 on Mint 17.3. Cloned and compiled libplist, libusrmuxd, libimobiledevice, usbmuxd, libirecovery, ideviceinstaller, idevicerestore from github masters today. 

`mark (master) libimobiledevice $ ideviceinfo`
`ERROR: Could not connect to lockdownd, error code -2`

but ideviceinfo -s works OK.

idevicebackup2 also fails with error code -2, even if I specify the UDID. 

iPhone 5S running iOS 10.3.3.

Alhadis, 2018-03-14:
Getting the same error here too. Anybody have any luck sorting this out?

walkjivefly, 2018-04-08:
I still get the same problem as reported on 17 Sept 2017. I recompiled all of libplist, libusbmuxd, libimobiledevice, usbmuxd, libirecovery, ideviceinstaller, idevicerestore from Github masters today. Linux Mint 17.3. iPhone 5S (ME298LL/A) with iOS 10.3.3 (14G60).

Reconfigured with debug, recompiled, using the -d option on idevicebackup2 as an example I get
```
mark $ idevicebackup2 -d -u a845f22dfdda19f0edad395348740a7ba6ed39c7 backup .
Backup directory is "."
20:55:34 lockdown.c:675 lockdownd_client_new(): device udid: a845f22dfdda19f0edad395348740a7ba6ed39c7
20:55:34 lockdown.c:405 lockdownd_query_type(): called
20:55:34 property_list_service.c:128 internal_plist_send(): sending 287 bytes
20:55:34 service.c:140 service_send(): sending 4 bytes
20:55:34 service.c:140 service_send(): sending 287 bytes
20:55:34 property_list_service.c:133 internal_plist_send(): sent 287 bytes
20:55:34 property_list_service.c:134 internal_plist_send(): printing 287 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>idevicebackup2</string>
        <key>Request</key>
        <string>QueryType</string>
</dict>
</plist>
20:55:34 property_list_service.c:193 internal_plist_receive_timeout(): initial read=4
20:55:34 property_list_service.c:202 internal_plist_receive_timeout(): 297 bytes following
20:55:34 property_list_service.c:215 internal_plist_receive_timeout(): received 297 bytes
20:55:34 property_list_service.c:241 internal_plist_receive_timeout(): printing 297 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Request</key>
        <string>QueryType</string>
        <key>Type</key>
        <string>com.apple.mobile.lockdown</string>
</dict>
</plist>
20:55:34 lockdown.c:421 lockdownd_query_type(): success with type com.apple.mobile.lockdown
20:55:34 property_list_service.c:128 internal_plist_send(): sending 335 bytes
20:55:34 service.c:140 service_send(): sending 4 bytes
20:55:34 service.c:140 service_send(): sending 335 bytes
20:55:34 property_list_service.c:133 internal_plist_send(): sent 335 bytes
20:55:34 property_list_service.c:134 internal_plist_send(): printing 335 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>idevicebackup2</string>
        <key>Key</key>
        <string>ProductVersion</string>
        <key>Request</key>
        <string>GetValue</string>
</dict>
</plist>
20:55:34 property_list_service.c:193 internal_plist_receive_timeout(): initial read=4
20:55:34 property_list_service.c:202 internal_plist_receive_timeout(): 327 bytes following
20:55:34 property_list_service.c:215 internal_plist_receive_timeout(): received 327 bytes
20:55:34 property_list_service.c:241 internal_plist_receive_timeout(): printing 327 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Key</key>
        <string>ProductVersion</string>
        <key>Request</key>
        <string>GetValue</string>
        <key>Value</key>
        <string>10.3.3</string>
</dict>
</plist>
20:55:34 lockdown.c:474 lockdownd_get_value(): success
20:55:34 lockdown.c:485 lockdownd_get_value(): has a value
ERROR: Could not connect to lockdownd, error code -2
```

walkjivefly, 2018-06-03:
Eventually found I had a distro-installed, wrong, version of one of the pre-reqs on my machine. I can't remember which one it was, possibly libplist or (lib)usbmuxd. apt/synaptic wouldn't let me remove the distro-supplied version without completely lobotomising the install. I ended up building a Linux From Scratch system to be sure it was free from crap and then compiling all the latest github masters. The result was a fully working libimobiledevice etc.

sc84647, 2021-12-10:
I was getting this error until I unlocked the device, got the Trust This Computer? popup, and chose Trust.