libimobiledevice/libimobiledevice #380: Release a new version

kirb, 2016-09-20:
The latest release of libimobiledevice is 1.2.0, released almost 2 years ago now. This release [no longer works](https://github.com/libimobiledevice/libimobiledevice/issues?utf8=%E2%9C%93&q=could%20not%20connect%20to%20lockdownd) with recent iTunes releases. Package managers tend to only release updates for a package when upstream does a release, especially Homebrew which heavily distinguishes between "released" (what you get by default) and "unreleased" (what you get by using `--HEAD`).

Is it possible that we could see a new release in the near future, and possibly, regular-ish releases following that?

ledinhphuong, 2016-09-28:
Yes, please

txaj, 2016-10-09:
+1

This also prevents Debian from packaging a working release.

(Duplicates #340)

TheElectronWill, 2016-10-17:
definitely +1 !

TheElectronWill, 2016-10-25:
@nikias Any plan to do it?

nikias, 2016-10-25:
@TheElectronWill yes soonish

jespinal, 2016-10-31:
+1

I'm trying to reach a recently updated (2016-10-30 20:45:00) iPhone 6s+ and get this on Ubuntu MATE 16.04.1:

GnuTLS error: Error in the pull function.
ERROR: Could not connect to lockdownd, error code -5

Unfortunately there's no package update until a new release is available.

TheElectronWill, 2016-12-25:
@nikias gentle reminder :-)

Frederick888, 2016-12-28:
Grrrrr.... the AUR version hasn't been update for quite long, either... +1...

TheElectronWill, 2017-01-23:
For those who need the lastest libimobiledevice, here are some helpful links:

- [M. Salbaba's Ubuntu PPA](https://launchpad.net/~martin-salbaba/+archive/ubuntu/ppa+libimobiledevice)
It isn't completely up to date with the upstream libimobiledevice, but it should be enough for working properly with iOS 10. (I haven't tested it since I don't use Ubuntu).
- [My script to build an RPM package of the upstream version](https://github.com/RPM-Outpost/libimobiledevice)
I've made a shell script that downloads the latest (unreleased) version of libimobiledevice and create a ready-to-use RPM package.

AlexJinsin, 2017-03-21:
#188 --HEAD can fix -21

jrmhaig, 2017-03-30:
Yes, `--HEAD` fixes it because that is installing from the current head on Github. This is generally not a good thing to do as you could end up with unexpected changes. However, in this case it is the only option until there is a new release.

samrocketman, 2017-04-27:
For those who can't wait for a release, I've published a guide of how I built libimobiledevice for myself on GNU/Linux.  Perhaps you may benefit.  https://gist.github.com/samrocketman/70dff6ebb18004fc37dc5e33c259a0fc

supermarin, 2017-05-26:
Seriously, please get this released

wbob, 2018-08-19:
including this one, I count 7 "new release" threads to date (#631 #585 #456 #405 #401 #340), being testament to the importance of the library. I just noticed [libimobiledevice.org](https://www.libimobiledevice.org/) has a donate button with a customizable message. Though I guess most current distributions will now package from a git commit, I think it's a way to show appreciation and make your voices heard.

FunkyM, 2020-06-15:
Closing as version [1.3.0](/libimobiledevice/libimobiledevice/releases/tag/1.3.0) has been released.

closed 2020-06-15 by FunkyM