libimobiledevice/libimobiledevice #382: iOS 10.0.1 not supported?

MirukaiYT, 2016-09-20:
I have been trying to use the libimobiledevice tools for a while today, but I can't seem to get them to work. When I pair them, it does it successfully, but then when I go on to other commands, `idevicebackup` for example, it fails, and gives me these errors:

`GnuTLS error: Error in the pull function.
ERROR: Could not connect to lockdownd, error code -5`

Can anyone help me with this?

qqtester, 2016-09-21:
I just fix this. But i am not quite sure why.
change openssl to 1.1.0, the new version. rebuild the project. you can try this.
1. ideviceinfo
2. sudo ideviceinfo
see what happened.
Now, my problem is some warning showed after execute command. as below:
`yuriwangdeiMac:libimobiledevice-master yuriwang$ idevice_id -l
83a416aaf399cb108fe3355a6e9cacda4963fdd2
idevice_id(17785,0x7fff77ef3000) malloc: *** error for object 0x7fcf09c07ea0: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Abort trap: 6`

MirukaiYT, 2016-09-21:
Ah, thanks @qqtester that worked! I appreciate the help!

qqtester, 2016-09-22:
@MirukaiYT , did you get the same warning as above mentioned?

inspectionsbybob, 2016-09-22:
I had been using libiMobileDevice 1.1.2-0.1 for quite some time with Linux Mint 17.3 and my iDevices.  Everything worked "acceptable" or had workarounds.
After I upgraded to iOS10 it BROKE.

I now get "Unable to mount [device name]" Unhandled Device lockdown error (-256)"

I have tried to compile the latest version (1.2) which is supposed to work with iOS10, but can't find a "complete" library that will build all the way through.

I am not a Linux guru, just an average person who really hates windows, so traipsing all over the internet to try to assemble all of the bits needed to compile 1.2 is beyond me. Give me a complete list of commands and I can follow instructions like anyone, gut "get the needed library" doesn't help me....

HELP Please, I hate having to boot into windows to pull my files off my iPad.

nikias, 2016-09-22:
1.2 is not the latest code available. You need to get the latest code from git master for iOS 10 support.

inspectionsbybob, 2016-09-22:
I tried downloading the library from libimobiledevice.org which should have the latest code.  Their Blog specifically says it works with iOS10 which would make me happy. The library is incomplete and won't compile (build).

Is there a Sudo-get or install command that will get me the latest version that will work on Linux Mint 17.3?

I have looked carefully for either a pre-compiled version, or a COMPLETE library that I could compile locally, and can't find anything (understandable to be)

MirukaiYT, 2016-09-22:
@Inspectionsbybob

What you should do is use the `sudo apt-cache search` trick and find the aptitude packages. That's what I did. Then you need to download and compile OpenSSL 1.*.*, which isn't compiled in the official repos. 
Use `git clone https://github.com/openssl/openssl.git && cd openssl && ls`

You should see files named

`autogen.sh`
And
`configure`

If you see `autogen.sh` run that, if you only see `configure` then do `./configure`.

Once you have run the compilers (AutoGen and Configure`, run

MirukaiYT, 2016-09-22:
`Make` followed by `sudo make install`

To validate your install, run

`openssl version`
And you should get an output similar to this:

`john@randompc ~/openssl $ openssl version
OpenSSL 1.1.2`

Now re-try any of the commands in the `imobiledevice` packages.

Hope this helped!

inspectionsbybob, 2016-09-23:
OK, I almost followed that

from what folder open a terminal session???

I open a terminal session
and type
git clone https://github.com/openssl/openssl.git && cd openssl && ls

then autogen.sh   or ./configure

make
sudo make install

openss1 version

then go to the libimobiledevice folder and 

autogen.sh
configure
{the comnands in the readme file)

inspectionsbybob, 2016-09-23:
If you wouldn't mind, long, simple directions and don't assume I know something.
If I need to open a terminal session while in a particular folder, please tell me...
step by step please...

MirukaiYT, 2016-09-23:
Okay, @inspectionsbybob you need to open Terminal.

You should do this:
1 - Place yourself in your Home directory (`cd ~`)
2 - Clone the OpenSSL Git Repo into a folder (`git clone https://github.com/openssl/openssl.git`)
3 - Place yourself in the newly-made directory "openssl" (`cd openssl/`)
4 - Run the AutoMake tool, which should run ./configure as well, so don't worry about that. (`sudo sh automake.sh`)
5 - Prepare the OpenSSL build for installing (`make`)
6 - Install the files to your `/usr/` directories (`sudo make install`)

If you have any issues doing this, please don't hesitate to ask me via my email. Please title it:
`(Issue #) - (Your username) - I need help!`
followed by whatever went wrong.

If you were given an issue such as `autogen: autoreconf not found` then you need to install `autoconf`, `autoconf-dev` and `cmake`.

inspectionsbybob, 2016-09-23:
(Issue #382) - (@inspectionsbybob) - I need help
Installed git (sudo apt-get install git) 
then -----------------------------------------------
git clone https://github.com/openssl/openssl.git
remote: Repository not found.
fatal: repository 'https://github.com/openssl/openssl.git/' not found 

What now?

Also, when do I 'make' libimobiledevice?

MirukaiYT, 2016-09-23:
That's odd, try searchingg for the OpenSSL repo on here, and clone it
through the link on there.

Also, you should 'make' the libimobiledevice library whenever you please.

inspectionsbybob, 2016-09-23:
(Issue #382) - (@inspectionsbybob) - I need help
Ok, got openssl cloned, got automake installed

now I get "configure.ac' is required'  when I type "automake'

is the failure of libimobiledevice and iOS10 a problem with openssl or with libimobildevice?

inspectionsbybob, 2016-09-23:
(Issue #382) - (@inspectionsbybob) - I need help

I removed the openssl directory, and started again...

bob@bob-Inspiron-580 ~ $ git clone https://github.com/openssl/openssl.git
Cloning into 'openssl'...
remote: Counting objects: 219108, done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 219108 (delta 2), reused 0 (delta 0), pack-reused 219095
Receiving objects: 100% (219108/219108), 94.62 MiB | 6.57 MiB/s, done.
Resolving deltas: 100% (170169/170169), done.
Checking connectivity... done.
Checking out files: 100% (6746/6746), done.
bob@bob-Inspiron-580 ~ $ cd openssl
bob@bob-Inspiron-580 ~/openssl $ sudo sh automake.sh
[sudo] password for bob: 
sh: 0: Can't open automake.sh
bob@bob-Inspiron-580 ~/openssl $ ls
ACKNOWLEDGEMENTS  Configurations  external         NEWS         README.ENGINE
apps              Configure       FAQ              NOTES.DJGPP  README.FIPS
appveyor.yml      CONTRIBUTING    fuzz             NOTES.PERL   ssl
AUTHORS           crypto          include          NOTES.VMS    test
build.info        demos           INSTALL          NOTES.WIN    tools
CHANGES           doc             LICENSE          os-dep       util
config            engines         Makefile.shared  README       VMS
config.com        e_os.h          ms               README.ECC
bob@bob-Inspiron-580 ~/openssl $

MirukaiYT, 2016-09-24:
Okay, gimme a while and I'll make a silent tutorial video for you.

MirukaiYT, 2016-09-24:
Or, if you want, we could use Google Hangouts. By reading my bio, you
probably already know I'm twelve though, so...

If you say yes, I will try and compile the two on my Linux laptop running
Mint 18 'Sarah'. (I think it's 18...)

inspectionsbybob, 2016-09-24:
I am being sent out today (we are trying to sell our house, so we have a showing) so hangout is not likely...
I am running
1.1.5+git20140313.bafe6a9e-0ubuntu1.1
Linux Mint 17.3 - 2.8.8 -  3.16.0-38-generic

I am pretty good at following instructions

assume that my system is BARE, and I haven't already downloaded any utilities, it is just the software that comes with the full distribution... like autoconf, or git, openssl, or....

walk me through all of the steps for a full compilation....

start with the fist terminal session, and include each sudo apt-get and git clone necessary

I have tried almost all of the scripts in the copies of the libimobiledevice folders on the web, and they all are incomplete somewhere, with some library missing, or forgetting to run some config.sh script before you run autoconf or autosomething else... or forgetting that you need the newest openssl before you work on libimobiledevice...

If you want me to check for certain libraries first, just let me know the command to test for it, and the sudo apt-get install (necessary command) if it is not there...

I routinely run update, so my 17.3 system SHOULD be as up to date as the system update knows about.

my software sources are set to 

Main (rosa) packages.linuxmint.com
Base (trusty) mirror.symnds.com/ubuntu

Source code [ ] Enable source code repositories                is -NOT- selected.

When I was twelve, it was Assembler (Z80) and Pascal, dial up modems and lots of other antique things. I am jealous of what you take for granted. I could give you the thoughts of an OLD coder, but that is for another day. 

So assume I know barely enough to get confused and frustrated, I know how it is SUPPOSED to work, now why isn't it doing what I want...

inspectionsbybob, 2016-10-01:
Haven't heard anything in a few days...

More and more people are having issues with iOS10.... so if someone builds a "complete" library or procedure, you will be very popular....

haydenbjyoung, 2016-10-04:
Hey All, I have remedied my own issue!

Head to my repo to clone it, I will now close the issue tracker. Please post any problems on my issues page on my repo.

https://github.com/haydenbjyoung/libimobile-installer

closed 2016-10-04 by haydenbjyoung

gitsop01, 2016-10-05:
ON LINUX:

One catch you have to look out for is if you are installing a library
not installed by the distribution package manager is that the package
manager will usually install system libraies in /usr/lib(32bit) or
/usr/lib64(64bit)
but local source compliled on the system will usually install them in
/usr/local/lib or /usr/local/lib64 as the installed directory. The
system will still used the first library it finds in the path and that
is usually the /usr/lib or /usr/lib64 unless modified.

To avoid this the local build script should search and remove the old
library first from  system's /usr/lib or /usr/lib64.

Then install the new compiled library in /usr/local/lib or
/usr/local/lib64 to ensure that it is this library that is used.

nikias, 2016-11-04:
The pairing record generation with newer GnuTLS was broken. I just fixed it with commit https://github.com/libimobiledevice/libimobiledevice/commit/23069d10341ce637fdad7321d447c53752dba48c.
Make sure to remove invalid paring records by deleting the appropriate files or call ```idevicepair unpair``` for the affected devices.