libimobiledevice/libimobiledevice #383: warning showed after execute command as ideviceinfo : pointer being freed was not allocated

qqtester, 2016-09-21:
`yuriwangdeiMac:libimobiledevice-master yuriwang$ idevice_id -l
83a416aaf399cb108fe3355a6e9cacda4963fdd2
idevice_id(17785,0x7fff77ef3000) malloc: *** error for object 0x7fcf09c07ea0: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Abort trap: 6`
I just update openssl to 1.1.0, and rebuild the libimobiledevice. After execute command idevice_id -l , I got the warning like that. how to fix this? All the command show the same warning.

nikias, 2016-09-24:
I don't think this is a bug in libimobiledevice. Seems to me that something weird happened during compilation? If you want to find out where it happened, you can run the tool in lldb:
```
llldb -- idevice_id -l
```
Then type
```
b malloc_error_break
r
```
And paste the output.

nikias, 2016-10-04:
@qqtester are you still seeing this problem? If so, please provide the above debug info. Otherwise I'll have to close this ticket.

qqtester, 2016-10-08:
`yuriwangdeiMac:~ yuriwang$ lldb -- idevice_id -l
(lldb) target create "idevice_id"
Current executable set to 'idevice_id' (x86_64).
(lldb) settings set -- target.run-args  "-l"
(lldb) b malloc_error_break
Breakpoint 1: where = libsystem_malloc.dylib`malloc_error_break, address = 0x0000000000012f2e
(lldb) r
Process 89245 launched: '/usr/local/bin/idevice_id' (x86_64)
idevice_id(89245,0x7fff7dddd000) malloc: *** error for object 0x100504e70: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Process 89245 stopped
* thread #1: tid = 0x3d94a0, 0x00007fff96febf2e libsystem_malloc.dylib`malloc_error_break, queue = 'com.apple.main-thread', stop reason = breakpoint 1.1
    frame #0: 0x00007fff96febf2e libsystem_malloc.dylib`malloc_error_break
libsystem_malloc.dylib`malloc_error_break:
->  0x7fff96febf2e <+0>: pushq  %rbp
    0x7fff96febf2f <+1>: movq   %rsp, %rbp
    0x7fff96febf32 <+4>: nop    
    0x7fff96febf33 <+5>: nopl   (%rax)
(lldb) `
showed as above.
PS: sorry for the late, a vocation for National Day.

qqtester, 2016-10-08:
@nikias , any idea for this?

mexmer, 2016-10-10:
unfortunately screenshot throws error 503

qqtester, 2016-10-11:
@mexmer ,like this: 
`idevice_id(58551,0x7fff74fd9000) malloc: *** error for object 0x7fe959d04ac0: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Abort trap: 6`

mexmer, 2016-10-11:
now i see screenshot also, guess it was temporary error, anyways, since it's refering leakage in com.apple.main-thread it's related to OS or compilation, not to libimobile, looks guys from cupertino don't have everything cleaned

might be even caused by specific version of Xcode compiler.

of course, i might be mistaken, i have not debugged much code on OS X, i'm mainly windows user.

you should debug idevice_id, and see where it stops/breaks.

nikias, 2016-10-23:
@qqtester after typing r to execute the program ib lldb and it stops at malloc_error_break, type: bt

qqtester, 2016-10-25:
the lib is working now. thx. this could be closed.

closed 2016-10-25 by qqtester

Gamadril, 2016-12-30:
a small explanation for people like me who found that issue when searching for a solution and still not know what to do...
nikias suggestion brought me to the problem I had.
bt shows following output:
```
* thread #1: tid = 0x1ad48b, 0x00007fff87654f2e libsystem_malloc.dylib`malloc_error_break, queue = 'com.apple.main-thread', stop reason = breakpoint 1.1
  * frame #0: 0x00007fff87654f2e libsystem_malloc.dylib`malloc_error_break
    frame #1: 0x00007fff87645fd2 libsystem_malloc.dylib`free + 314
    frame #2: 0x00000001001ba836 libcrypto.1.1.dylib`OPENSSL_sk_pop_free + 84
    frame #3: 0x000000010003dd51 libssl.1.1.dylib`ssl_library_stop + 41
    frame #4: 0x0000000100167ed7 libcrypto.1.1.dylib`OPENSSL_cleanup + 101
    frame #5: 0x00007fff87853451 libsystem_c.dylib`__cxa_finalize_ranges + 327
    frame #6: 0x00007fff87853767 libsystem_c.dylib`exit + 55
    frame #7: 0x00007fff96a3d5b4 libdyld.dylib`start + 8
```

looking at the OPENSSL call I realized that the libimobiledevice library is linked against wrong openssl library version (1.1 in my case). After rebuilding libimobiledevice and linking against openssl 1.0 the error disappeared