libimobiledevice/libimobiledevice #386: libimobiledevice not working on iOS 10 device

darivb, 2016-09-23:
I have a project which uses the libimobiledevice library (installed using brew) and it used to work on my ios 9 device. Lately I upgraded my device to iOS 10 and now I can't use any of libimobiledevice binaries anymore, they all fail on "Could not connect to lockdownd".

For example:
$ idevicescreenshot
returns: ERROR: Could not connect to lockdownd, error code -5

$ ideviceinstaller 
returns: Could not connect to lockdownd. Exiting.

$ ifuse 
returns: Failed to connect to lockdownd service on the device. Try again. If it still fails try rebooting your device.

First question is what is lockdownd service exactly?
Second, what I do to make it work?

hexenesi, 2016-09-23:
I was having the same issue, and somewhere here in the Issues I find the solution:
Do the following:
brew uninstall libimobiledevice
then:
brew install --HEAD libimobiledevice

that should work.

inspectionsbybob, 2016-09-23:
"No command 'brew' found, did you mean:"

So my system must not have the required command

hexenesi, 2016-09-23:
@inspectionsbybob 
The solution is for MacOSX, not for Linux.

darivb, 2016-09-24:
thanks a lot @hexenesi , it worked!

closed 2016-09-24 by darivb