libimobiledevice/libimobiledevice #387: IOS 10.0.2 : Failed to connect to lockdownd service on the device

OSS542, 2016-09-25:
bash-3.2$ ifuse /mnt/iPhone/
Failed to connect to lockdownd service on the device.

bash-3.2$ ideviceinfo
ERROR: Could not connect to lockdownd, error code -5

ideviceinfo -s works

[jfh1.txt](https://github.com/libimobiledevice/libimobiledevice/files/491721/jfh1.txt)

currently installed under linux:
libimobiledevice-1.2.0
ifuse-1.1.3
libplist-1.12
libusbmuxd-1.0.10
usbmuxd-1.1.0

OSS542, 2016-09-25:
additional note: the latest clone from the repository fixes this issue.

gitsop01, 2016-09-25:
I can confirm this on IOS 9.3.2 also using the 1.20 standard released libmobiledevice llibrary suite
but have not tried with the git master head with the openssl source patch as listed on other issues numbers.
Also confirmed on IOS 9.3.2 that ideviceinfo -s works.

This is using Linux on Fedora 24 workstation updated with the latest stable packages.

Have you tested on latest master head clone on Linux with an older idevice IOS like 5.30 or 6.10 to see if it still works as expected.

OSS542, 2016-09-25:
I have not.  I only have the more recent releases of IOS available..

tvjon, 2016-09-26:
I usually download the zips rather than clone, but just on the off chance having read OSS542´s post , I cloned  libimobiledevice, but exactly the same lockdown error for 10.x.x
	
On an older iPad which can´t be upgraded to 10, & has 9.3.5, it mounts fine in Raspbian.

I built several versions, --disable-opennssl, so using gnutls, & the built in version, but still the lockdown errors persist.

OSS542, 2016-09-26:
some additional notes:
"libimobiledevice" was cloned from git and installed on an iPhone 5c running IOS 10.0.2.  The default configuration for libimobiledevice was used, with no configuration switches added.
The Linux used is entirely built from source (not LFS) using a custom designed procedure.

tvjon, 2016-09-26:
Thank you for the extra notes.
	
It looks like my problem was in fact, replicated libraries in:

/usr/lib/arm-linux-gnueabihf

These had 2014 dates on them, but I don´t know how they were originally installed.
I removed them, rebuilt the relevant libraries & now I can mount 10.0.2 devices via ifuse.

closed 2016-09-27 by nikias