libimobiledevice/libimobiledevice #388: Can not get device color and device ensure color on iOS 10

truonggiang0710, 2016-09-26:
I have an iPhone 5 (5,2) and iPhone 7 (9,3).
I use libimobiledevice (last version) to read information. But I can get color/ color hex of them.
Please help me!

aymenim, 2016-09-26:
@truonggiang0710 please be specific if the device color is what you are looking for ```ideviceinfo```  will give you ``` DeviceColor: black ``` something like this

truonggiang0710, 2016-09-27:
@aymenim  looke like as your mention on iPhone 5. But on iPhone 7, I can not get anything !

nikias, 2016-09-27:
@truonggiang0710
```
$ ideviceinfo -k DeviceColor
1
```
It seems the value is numeric for the iPhone 7.

truonggiang0710, 2016-09-27:
@nikias I think you right. Because I get 2 on iPhone 7 rose and iPhone 7 plus rose. I will update more value soon.

nikias, 2016-10-04:
In any case, this is not a bug in libimobiledevice. Please track the values elsewhere.

closed 2016-10-04 by nikias

truonggiang0710, 2016-10-26:
@nikias You are right.
I update more value of color.( Value of Device Ensure Color)
Black: 1
Silver: 2
Gold: 3
Jet Black: 4

Thanks all for your support.

oznotes, 2020-07-22:
// if iPhone 7 matte.
                {"1","Black"},
                // if Not silver model it is white.
                {"2","White"},
                {"3","Gold"},
                {"4","Rose Gold"},
                {"5","Jet Black"},
                {"6","Red"},
                {"7","Yellow"},
                {"8","Coral"},
                {"9","Blue"},

iwannet, 2024-04-12:
More updated list:

def get_color():
    try:
        output = subprocess.check_output(['ideviceinfo', '-k', 'DeviceEnclosureColor'])
        color_code = output.decode().strip()
        color_mapping = {
            '#3b3b3c': 'Zwart',
            '#ffffff': 'Wit',
            '#ff3b30': 'Rood',
            '#ff9500': 'Oranje',
            '#ffcc00': 'Geel',
            '#4cd964': 'Groen',
            '#5ac8fa': 'Blauw',
            '#007aff': 'Lichtblauw',
            '#5856d6': 'Paars',
            '#ff2d55': 'Roze',
            '#8e8e93': 'Grijs',
            '#c69c6d': 'Goud',
            '#d0d1d2': 'Zilver',
            '1': 'Zwart',
            '2': 'Wit',
            '3': 'Goud',
            '4': 'Roze',
            '5': 'Grijs',
            '6': 'Rood',
            '7': 'Geel',
            '8': 'Oranje',
            '9': 'Blauw',
            '17': 'Paars',
            '18': 'Groen',
            
        }
        return color_mapping.get(color_code, 'Unknown Color')
    except subprocess.CalledProcessError:
        return "No Device Connected"