libimobiledevice/libimobiledevice #389: 'openssl/ssl.h' file not found

aroramayank2002, 2016-09-26:
While trying to build libimobiledevice after git clone
make command failed with this exception:

/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in common
  CC       debug.lo
In file included from debug.c:36:
../src/idevice.h:30:10: fatal error: 'openssl/ssl.h' file not found
#include <openssl/ssl.h>
         ^
1 error generated.
make[2]: *** [debug.lo] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

Seems the solution is to add openssl include path to make configuration as Apple has depcricated use of openssl in ElCapitan but I don't know how to give the parameter to make file.

on "brew info openssl" we get

Apple has deprecated use of OpenSSL in favor of its own TLS and crypto libraries

Generally there are no consequences of this for you. If you build your
own software and it requires this formula, you'll need to add to your
build variables:

    LDFLAGS:  -L/usr/local/opt/openssl/lib
    CPPFLAGS: -I/usr/local/opt/openssl/include
    PKG_CONFIG_PATH: /usr/local/opt/openssl/lib/pkgconfig

May be its not an issue with libimobiledevice but can someone help me with adding cpp flags to make file.

nikias, 2016-09-27:
Your openssl version is too old if it doesn't have ERR_remove_thread_state.

aroramayank2002, 2016-10-07:
I updated openSSL following this thread:
 http://stackoverflow.com/questions/38670295/brew-refusing-to-link-openssl

openssl version
OpenSSL 1.0.2i  22 Sep 2016

And did: ./autogen.sh and make but same error.

jpsalada, 2016-10-07:
Same here.
**MacOS Version:** macOS 10.12 
**Opensll Version:**OpenSSL 1.0.2j  26 Sep 2016

**Configure Command:** ./configure openssl_CFLAGS=-I/usr/local/opt/openssl/include openssl_LIBS=/usr/local/opt/openssl/lib/ PKG_CONFIG_PATH=/usr/local/opt/openssl/lib/pkgconfig

**Make result:**

/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in common
  CC       socket.lo
  CC       thread.lo
  CC       debug.lo
  CC       userpref.lo
  CC       utils.lo
  CCLD     libinternalcommon.la
Making all in src
  CC       idevice.lo
  CC       service.lo
  CC       property_list_service.lo
  CC       device_link_service.lo
  CC       lockdown.lo
  CC       afc.lo
  CC       file_relay.lo
  CC       notification_proxy.lo
  CC       installation_proxy.lo
  CC       sbservices.lo
  CC       mobile_image_mounter.lo
  CC       screenshotr.lo
  CC       mobilesync.lo
  CC       mobilebackup.lo
  CC       house_arrest.lo
  CC       mobilebackup2.lo
  CC       misagent.lo
  CC       restore.lo
  CC       diagnostics_relay.lo
  CC       heartbeat.lo
  CC       debugserver.lo
  CC       webinspector.lo
  CC       syslog_relay.lo
  CCLD     libimobiledevice.la
Undefined symbols for architecture x86_64:
  "_ASN1_INTEGER_free", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_ASN1_INTEGER_new", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_ASN1_INTEGER_set", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_ASN1_TIME_free", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_ASN1_TIME_new", referenced from:
(Errors list goes on..)

qqtester, 2016-10-10:
face the same error. anyone fix this?

n3il, 2016-10-19:
I think the best solution is settings environment variables like the message below. 

```
> brew info openssl
...
Apple has deprecated use of OpenSSL in favor of its own TLS and crypto libraries

Generally there are no consequences of this for you. If you build your
own software and it requires this formula, you'll need to add to your
build variables:

    LDFLAGS:  -L/usr/local/opt/openssl/lib
    CPPFLAGS: -I/usr/local/opt/openssl/include
```

I did manage to get it working with `brew link --force openssl`

cffls, 2016-10-27:
So I was able to solve this problem by making a symbolic link on my Mac. 
Here is my openssl.pc:
```
prefix=/usr
exec_prefix=${prefix}
libdir=${prefix}/local/Cellar/openssl/1.0.2j/lib
includedir=${prefix}/local/Cellar/openssl/1.0.2j/include

Name: OpenSSL
Description: Secure Sockets Layer and cryptography libraries and tools
Version: 1.0.2j
Requires: 
Libs: -L${libdir} -lssl -lcrypto -lz
Cflags: -I${includedir}
```
However, Mac seems to continue using /usr/local/lib/libssl.a
So I got this error:
```
Making all in src
  CCLD     libimobiledevice.la
ld: warning: ignoring file /usr/local/lib/libssl.a, file was built for archive which is not the architecture being linked (x86_64): /usr/local/lib/libssl.a
Undefined symbols for architecture x86_64:
```

So I simply made a symbolic link by:
```
mv /usr/local/lib/libssl.a /usr/local/lib/libssl-origin.a 
ln -sfv /usr/local/lib/libssl.a /usr/local/Cellar/openssl/1.0.2j/lib/libssl.a
```

And final I can compile the source without errors.

rpplusplus, 2016-11-01:
I have same problem and i disable openssl and use libgnutls instead

just
```
brew install gnutls 
brew install libgcrypt
./configure --disable-openssl
```

samrocketman, 2016-11-02:
Before building, try setting.

```bash
LD_LIBRARY_PATH=/usr/local/opt/openssl/lib:"${LD_LIBRARY_PATH}"
CPATH=/usr/local/opt/openssl/include:"${CPATH}"
PKG_CONFIG_PATH=/usr/local/opt/openssl/lib/pkgconfig:"${PKG_CONFIG_PATH}"
export LD_LIBRARY_PATH CPATH PKG_CONFIG_PATH
```

`CPATH` and `PKG_CONFIG_PATH` are for compiling.  `LD_LIBRARY_PATH` is for runtime.  See also https://gist.github.com/samrocketman/70dff6ebb18004fc37dc5e33c259a0fc

jpsalada, 2016-11-11:
@rpplusplus That worked for me. thanks

xutianxi, 2016-12-21:
@samrocketman thanks, it worked for me.

jekkt94, 2017-02-16:
@samrocketman thank you for your solution it worked for me! :)

inancgumus, 2017-03-26:
For homebrew, before building, try:

```bash
export LD_LIBRARY_PATH=$(brew --prefix openssl)/lib
export CPATH=$(brew --prefix openssl)/
export PKG_CONFIG_PATH=$(brew --prefix openssl)/lib/pkgconfig
```

Thanks @samrocketman

VasilijSviridov, 2017-06-14:
Hi,

Might you can help me, after running make I am getting: 

/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in common
make[2]: Nothing to be done for `all'.
Making all in src
  CCLD     libimobiledevice.la
ld: library not found for -lssl
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libimobiledevice.la] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

Any help ? 
Cheers

cookiesgreat, 2017-10-30:
@inancgumus It works ! Tested with High Sierra

khoahuynh47, 2017-12-18:
Thanks @inancgumus

YummyTastyCode, 2018-10-21:
Due to this is the very first link in the search engine I have to write my solution

```
ln -s /usr/local/Cellar/openssl/1.0.2o_1/include/openssl /usr/local/include/openssl
ln -s /usr/local/Cellar/openssl/1.0.2o_1/lib /usr/local/lib/openssl
```

ZhaoChuyang, 2019-04-11:
> I think the best solution is settings environment variables like the message below.
>
> ```
> ...
> Apple has deprecated use of OpenSSL in favor of its own TLS and crypto libraries
>
> Generally there are no [...]

it works, thank you~

kebbbnnn, 2019-07-12:
@YummyTastyCode that worked for me. thanks! ❤️

baijiahei, 2020-02-08:
brew reinstall openssl
export LDFLAGS="-L/usr/local/opt/openssl@1.1/lib"
export CPPFLAGS="-I/usr/local/opt/openssl@1.1/include"
export PKG_CONFIG_PATH="/usr/local/opt/openssl@1.1/lib/pkgconfig"
echo 'export PATH="/usr/local/opt/openssl@1.1/bin:$PATH"' >> ~/.zshrc
it works on my Mac