libimobiledevice/libimobiledevice #393: Fixed "It does not support ios10, lockdown error -5, how do? #327", BUT idevicebackup2: plist.c:238: plist_copy_node: Assertion `data' failed.

NormanFew, 2016-10-01:
Recompiled with the git head and ideviceinfo worked for iOS10, but idevicebackup2 failed with " idevicebackup2: plist.c:238: plist_copy_node: Assertion `data' failed.". 

Any idea? Thanks for help.

NormanFew, 2016-10-04:
The issue I ran into was caused by connecting to an iPad not with a CDMA module.
I modified idevicebackup2.c at line #401as
value_node = plist_dict_get_item(root_node, "MobileEquipmentIdentifier");
#if 1 //device without CDMA
	if (value_node)
		plist_dict_set_item(ret, "MEID", plist_copy(value_node));
#else
	plist_dict_set_item(ret, "MEID", plist_copy(value_node));
#endif

nikias, 2016-10-04:
Fixed with commit https://github.com/libimobiledevice/libimobiledevice/commit/1fa41702b6c7b8b169a33b79125d3b24f23c2da8

closed 2016-10-04 by nikias