libimobiledevice/libimobiledevice #394: Memory leak on OSX

sinosoidal, 2016-10-03:
Hi,

I'm having a memory leak on OSX. It goes up to 256 MB in memory in one hour.

On last version the problem was on libplist. I had to comment some parts to make it avoid the leak. It seems that the problem remains but that fix is not valid anymore because there were some structural modifications to the code. Does anyone knows anything about this?

Regards

nikias, 2016-10-03:
Can you be more specific? Which tools that use libimobiledevice are leaking memory?

sinosoidal, 2016-10-03:
Yes, of course. 

I use libimobiledevice to be able to establish wired communication between ios device and computer.

First of all I subscribe the events to know when a device is added or removed:

    idevice_event_subscribe(device_event_cb, this);

When a new device is added I lock it down and if connected successfully I start polling it:

    idevice_new(&_device, _udid.toLocal8Bit().constData())
    lockdownd_error_t lerr = lockdownd_client_new_with_handshake(_device, &_lockdown, "APP_NAME");

    if (lerr!=LOCKDOWN_E_SUCCESS)
    {
        idevice_free(_device);
        return;
    }

    char* name = NULL;
    lerr = lockdownd_get_device_name(_lockdown, &name);

    if (name)
    {
        _name = QString::fromUtf8(name);
        free(name);
    }
    else
    {
        qDebug("ERROR: Could not get device name, lockdown error %d\n", lerr);
    }

    lockdownd_client_free(_lockdown);

Then I poll it:

    idevice_error_t r;

    if (!_connected)
    {
        r = idevice_connect(_device,(uint16_t) WIRED_CONNECTION_PORT_IOS, &_connection);

        if (r!=IDEVICE_E_SUCCESS)
            return;

        _connected = true;

        emit connected();
    }
    else
    {
        uint32_t receivedBytes=0;
        char buffer[4096];

        r = idevice_connection_receive_timeout(_connection, buffer, sizeof(buffer), &receivedBytes, 1);

        if (r!=IDEVICE_E_SUCCESS)
            return;

        if (receivedBytes>0)
            _buffer.append(QByteArray(buffer, receivedBytes));
    }

I don't even need to pool it, without a device connected the memory also increases.

nikias, 2016-10-04:
So you're saying, a simple program that subscribes for device events (using idevice_event_subscribe) and you never connect any device, then it will leak memory?

sinosoidal, 2016-10-09:
Yes, but I double check that properly and get back with feedback. I'm not sure if it is enough to have events subscribed or if I need to have a device connected. I will get back with that info. Regards

nikias, 2016-10-21:
Well I do see memory usage increasing indeed. Some seems to come from libxml (via libplist). Some from OpenSSL. For libplist, I have a fix; I ripped libxml2 out of it. Will publish this soon.

sinosoidal, 2016-10-25:
That’s awesome! Thanks for checking this. I hadn’t time to get back to it.

Best!

nikias, 2016-10-25:
https://cgit.libimobiledevice.org/libplist.git/log/?h=time64

sinosoidal, 2016-10-25:
How can I test this?

I have git pull to libplist project and it is updated.

Any special procedure?

Thx,

Regards,

Nuno

nikias, 2016-10-26:
```
git fetch
git checkout time64
```

sinosoidal, 2016-10-27:
At /Users/nsantos/github/libplist

I made:

git fetch
git checkout time64

Result:

➜  libplist git:(master) git fetch
➜  libplist git:(master) git checkout time64
error: pathspec 'time64' did not match any file(s) known to git.

Ideas?

Nuno Santos
Founder / CEO / CTO
www.imaginando.pt
+351 91 621 69 62

nikias, 2016-10-28:
Make sure to check out via libimobiledevice.org. Just clone it in a new folder:

```
http://git.libimobiledevice.org/libplist.git
git fetch
git checkout time64
```

sinosoidal, 2016-10-29:
Nikias,

I have updated and recompiled my app. It seems to have radically improved. I have run my app for two hours and the memory just increased only 15 Mb. There is still a memory leak. I need to investigate if it isn’t on my side. 

Regards,

Nuno

nikias, 2016-10-31:
Awesome! I updated the branch with some bug fixes contributed by @filippobiga. I think the other memory leaks probably come from OpenSSL. You should be able to get the latest fixes by calling ```git pull```.

sinosoidal, 2016-11-08:
@nikias, any plans to merge this into the master branch? Should I use this branch for a release? Thanks!

nikias, 2017-04-26:
@sinosoidal do you have any update on further memory leaks?

sinosoidal, 2017-04-26:
I haven't mean noticed any. Has this been merged into master already? If so, I can compile the master and use the latest version of the lib to see if it is happening.

jflow, 2017-06-15:
Anyone know if this is on the master branch yet or not? If not any timeline for getting this fix moved over?

nikias, 2017-06-15:
@jflow @sinosoidal yes it's in master of libplist and has been released with libplist 2.0.0 already. If nothing too serious occurs anymore for you I'd close the ticket.

jflow, 2017-06-16:
Thank for the update. Will test and get back to you.

nikias, 2019-07-07:
Assuming this issue is actually resolved by now.

closed 2019-07-07 by nikias