libimobiledevice/libimobiledevice #396: sbservices_get_icon_state plist for WebClip provides bad URL

DaKaZ, 2016-10-05:
When calling sbservices_get_icon_state() the resulting PLIST seems to have an error or omission OR I just don't know how to properly retrieve the data.  The applications and folder data is all as expected, but for WebClips, the `webClipURL` key is always `/`.  For example:

```
2016-10-05 10:54:08.262790 K---[3341:1356667] displayName - ESPN: The Worldwide Leader in Sports
2016-10-05 10:54:08.262816 K---[3341:1356667] iconModDate - 2016-10-05 16:53:40 +0000
2016-10-05 10:54:08.262826 K---[3341:1356667] displayIdentifier - 8742BD3A648D44AEBD5E27BD013A64AC
2016-10-05 10:54:08.262834 K---[3341:1356667] webClipURL - /
```
and 
```
2016-10-05 11:14:54.303350 K---[3341:1356667] displayName - Fox News - Breaking News Updates | Latest News Headlines | Photos & News Videos
2016-10-05 11:14:54.303366 K---[3341:1356667] iconModDate - 2016-10-05 17:13:22 +0000
2016-10-05 11:14:54.303375 K---[3341:1356667] displayIdentifier - 43CAC393A3B94528BDB7469563170093
2016-10-05 11:14:54.303482 K---[3341:1356667] webClipURL - /
```
and I am logging the plist like this:
```
    for (id key in [dictionary allKeys]) {
        NSLog(@"%@ - %@",key,[dictionary objectForKey:key]);
    }
```

I am hoping this is a user error issue because I cannot find anything in the source code that would cause this problem.  Any ideas??

mexmer, 2016-10-05:
how you can be sure it is incorrect?
try with debug mode, so you will see raw communication, but i just checked some icons on my ipad with ios 10, and they show same behavior, none of them has webclipurl, that points to anything.

DaKaZ, 2016-10-05:
Hi @mexmer 

I cannot be sure.  Interestingly, if you look at the Home Screen Layout payload type from Apple's documentation they have recently removed the WebClip format (https://developer.apple.com/library/prerelease/content/featuredarticles/iPhoneConfigurationProfileRef/Introduction/Introduction.html).  The say:

> This payload defines a layout of apps, folders, and web clips for the Home screen. It is supported on iOS 9.3 and later.

But then the webClip TYPE is omitted from the dictionary format (you'll have to believe me that it used to be there!)

So maybe this is intended functionality.  I did test with a 9.3 device and it too reports the URL as just a `/` which is very odd.  I can't figure out why they would change this (or even if they did)

mexmer, 2016-10-05:
like with everything, because they can. they only need to be compatible with themselves.

i can see that, everytime they upgrade major ios version, and sometimes minor

nikias, 2016-10-05:
Keep in mind that sbservices is intended to provide a way to rearrange the icon layout. There's no point in getting the URL just to be able to rearrange the icons. Apple changed/removed it because it's not needed. The webclips are part of the device backup though (or at least used to be).
This is not a bug in libimobiledevice.

closed 2016-10-05 by nikias

DaKaZ, 2016-10-05:
Okay - thanks all, we'll just assume we no longer need the URL to set the home screen layout/payload.  Thanks for all the efforts, its greatly appreciated!!