libimobiledevice/libimobiledevice #397: warnings showed during build the project

qqtester, 2016-10-10:
`yuriwangdeiMac:libimobiledevice yuriwang$ make
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in common
  CC       debug.lo
  CC       userpref.lo
  CCLD     libinternalcommon.la
Making all in src
  CC       idevice.lo
idevice.c:105:2: warning: 'ERR_remove_thread_state' is deprecated [-Wdeprecated-declarations]
        ERR_remove_thread_state(NULL);
        ^
/usr/local/openssl/include/openssl/err.h:246:25: note: 'ERR_remove_thread_state' has been explicitly marked deprecated here
DEPRECATEDIN_1_1_0(void ERR_remove_thread_state(void *))
                        ^
idevice.c:706:33: warning: 'TLSv1_method' is deprecated [-Wdeprecated-declarations]
        SSL_CTX *ssl_ctx = SSL_CTX_new(TLSv1_method());
                                       ^
/usr/local/openssl/include/openssl/ssl.h:1596:45: note: 'TLSv1_method' has been explicitly marked deprecated here
DEPRECATEDIN_1_1_0(__owur const SSL_METHOD *TLSv1_method(void)) /* TLSv1.0 */
                                            ^
idevice.c:760:2: warning: 'ERR_remove_thread_state' is deprecated [-Wdeprecated-declarations]
        ERR_remove_thread_state(NULL);
        ^
/usr/local/openssl/include/openssl/err.h:246:25: note: 'ERR_remove_thread_state' has been explicitly marked deprecated here
DEPRECATEDIN_1_1_0(void ERR_remove_thread_state(void *))
                        ^
idevice.c:55:13: warning: unused function 'locking_function' [-Wunused-function]
static void locking_function(int mode, int n, const char* file, int line)
            ^
idevice.c:63:22: warning: unused function 'id_function' [-Wunused-function]
static unsigned long id_function(void)
                     ^
5 warnings generated.
  CC       service.lo`

any idea for this? guys.

mexmer, 2016-10-10:
deprecated methods, that were replaced with new ones in new version of SSL.
it will need code update, but for now, you can ignore them.

qqtester, 2016-10-10:
@mexmer ,ok. I got it. thx. 
could you take a look at #383 , after build success. run idevice_id -l. the message showed as the screenshot.

closed 2016-10-10 by nikias