libimobiledevice/libimobiledevice #399: ios10: the printf/NSLog message not output using idevicesyslog

suzhenghao, 2016-10-14:
there are  some log message output from my app, but all come from libsystem_network.dylib or UIKit. 
I use NSLog/printf to print some message, but none of them were printed by idevicesyslog.
can anyone help me ?
Thanks.

nikias, 2016-10-14:
This happens with Xcode too. This is not a bug. Apple changed the logging API with iOS 10.

suzhenghao, 2016-10-16:
@nikias  thank you very much. So is there any way to solve this problem?

jackrex, 2016-10-24:
i have same problem
i disable/enable this but not work

[image]

suzhenghao, 2016-10-26:
@jackrex 
I find something here:
1. use Xcode 8 build:
if use Xcode8 release version, NSLog will work.
2. use Xcode 7.3 build and submit the app to apple store, then use ios10 device to download the app, the NSLog doesn't work.

nikias, 2016-11-02:
I need to make clear that the NSLog/printf output from apps run through Xcode use the stdout from the remote debugger session. They don't use syslog relay at all.

argp, 2016-12-16:
Without running apps through Xcode, I don't get NSLog output with Xcode 8.2 and a device with iOS 10.1.1.

Are there any updates to this issue?