libimobiledevice/libimobiledevice #402: On Backup

Mradr, 2016-10-17:
Could not remove 'iPhoneBackups/UUID/Snapshot': Bad address (14)
ErrorCode 104: Error removing snapshot directory (MBErrorDomain/104). Underlying error: Bad address (MBErrorDomain/100). with this build when making the backup.﻿

Keep getting this error on backing up the device:
idevicebackup2 backup iPhoneBackups --full

schuylerdaddy, 2016-10-17:
I'm getting this exact same issue with an iPhone6, running on ios 10.0.1.

I am running a similar command on Windows 7: 

[image]

[image]

schuylerdaddy, 2016-10-17:
I'll also add here that the all the files seem to copy over just fine, it just seems like it throws an error for trying to delete that snapshot directory.

saeen7, 2016-10-17:
It seems like something not right with the file idevicebackup2. What I would do is download [this](https://cgit.libimobiledevice.org/libimobiledevice.git/snapshot/libimobiledevice-6aecc7228dbb30b4db48374ffb4ae4a1616bdd95.tar.gz) file, replace it with the current one and try again.

nikias, 2016-10-17:
Yeah, you need to use the latest code from git. However there's another bug where it fails to delete some files too and aborts (somehow the device requests to delete some files multiple times, first attempt works but on the second attempt it will report an error so the backup fails). I am working on a solution.

mexmer, 2016-10-18:
tbh. my code just logs problems with deleting files, folders, but does report success back.
there is postcleanup process that clears stuff anyway, so it would not cause problems, when starting new backup.
but when i was going trough logs i didn't see any duplicate deletion.

nikias, 2016-10-18:
@mexmer I understand that, however imagine - for whatever reason - the deletion of a file or folder fails for real. Then the backup might not be able to continue properly; also when performing a new backup the device might complain that the backup folder does contain files that are not in the backup database.

mexmer, 2016-10-18:
@nikias that's why there is postcleanup procedure in my program, it will always delete snapshot folder, if it still exists. if that fail, then program will generate error.

btw. backup from phone asks for directory listing at begining, and if snapshot folder exists, it asks to remove it first.

i understand your concerns tho'

Mradr, 2016-10-19:
Anyway I can get someone to make a new build for Windows then?

nikias, 2016-10-21:
@mexmer I actually didn't even see that it is basically trying to remove a file that is not present anymore. So I am getting ENOENT. So with my latest commit https://github.com/libimobiledevice/libimobiledevice/commit/2c16751bafb41a049e30f211c2a21c3b8c68259e I only report an error back to the device when something else than "file doesn't exist" happens.
@schuylerdaddy can you try again with the latest code?

nikias, 2016-10-21:
oh, actually @schuylerdaddy you're on windows, can you add this printf to see what the underlying win32 error code is? That would help I believe.
```
diff --git a/tools/idevicebackup2.c b/tools/idevicebackup2.c
index d355200..2be0314 100644
--- a/tools/idevicebackup2.c
+++ b/tools/idevicebackup2.c
@@ -203,6 +203,7 @@ static int win32err_to_errno(int err_value)
                case ERROR_ALREADY_EXISTS:
                        return EEXIST;
                default:
+printf("GOT WIN32 ERROR: %d\n", err_value);
                        return EFAULT;
        }
 }
```

schuylerdaddy, 2016-10-21:
@nikias I've actually compiled with the latest source code from libimobiledevice,libusbmuxd, and libplist on cyqwin (32-bit).  I have not been receiving this error anymore.  I think the previous executable I was using might have been outdated.  I didn't actually compile it, but I downloaded precompiled binaries from https://goo.gl/XzjWDy which were supposed to be compiled on Oct 4, 2016 by iDevice Central.

So in short, it is not creating this error currently on windows 7 with the latest build from source code.

schuylerdaddy, 2016-10-21:
On a further note, I can provide a step by step text file that says how to compile on a Windows machine using Cygwin.  It does seem many people ask about this and I would be happy to provide you with that if you want.

nikias, 2016-10-21:
@mexmer yep, that was my thought :)
Closing since the actual problem isn't a problem anymore.

closed 2016-10-21 by nikias