libimobiledevice/libimobiledevice #404: Backup is slow on ARMv7 device

zgoda-mobica, 2016-10-19:
This is rather general question.

I'm doing backup on ARMv7 device running Debian (dual core Alwinner A20 1.2GHz, 1GB RAM, gigabit ethernet, Class 10 SD card storage, 2xUSB2.0 host), using code from git HEAD. Backup of 2GB data takes ~4 minutes (compared to ~1 minute on my oldish Core i5 laptop running Ubuntu 16.04) and I'm looking to speed things a bit. Tried to store backup data on SD card and on NFSv3 storage connected by gigabit ethernet but this did not change much, as if the process was limited by something else than storage. Is it data reading from iphone slow, or the processor has not enough processing power?

Could someone explain me what's happening there and maybe what could I do to make this process a bit faster?

mexmer, 2016-10-19:
might be either cpu or usb host adapter issue (driver optimization). check cpu usage when doing backup.

nikias, 2016-10-19:
Yeah. CPU usage might be a good indicator. Also, you can try to increase the buffer size in idevicebackup2.c:894  mb2_handle_receive_files()
```
-	char buf[32768];
+	char buf[262144];
```
This might increase the throughput a little bit aswell, especially for large files.

zgoda-mobica, 2016-10-19:
Thank you for these hints. I monitored the process with vmstat. CPU utilization usually does not exceed 40% on single core (with spikes on SY when data are written), and uses only one of them. Backup on SD card shows some IO wait states but on NFS no such thing can be observed (WA). Attached is vmstat log from backup to NFS.

[vmstat-nfs1.log.txt](https://github.com/libimobiledevice/libimobiledevice/files/539602/vmstat-nfs1.log.txt)

nikias, 2016-10-19:
@zgoda-mobica have you changed the buffer size to see if that improves the speed (at least a little)?

zgoda-mobica, 2016-10-20:
With 256KB buffer size backup to SD card took longer, bit over 5 minutes for 2 GB data compared to ~4 minutes with original 32KB. There was less IO wait states but they was much longer and they was visible even on idevicebackup2 progress. I yet have to check with NFS share.

mexmer, 2016-10-20:
you might also try smaller one, like 16kB, still might be problem with ARM kernel optimization.

zgoda-mobica, 2016-10-20:
I made backup marginally faster by using std POSIX `open()` / `read()` / `write()` in `mb2_copy_file_by_path` with small buffer (8KB) to make copied data fit in L1 CPU cache but this was like 8-10 seconds overall gain.

nikias, 2016-10-20:
Hmm. Maybe the usb driver is just slow? The libimobiledevice code shouldn't unnecessarily slow things down, at least I am not aware of that. Same applies to usbmuxd and libusbmuxd. You can also make another test via AFC, put a large file (like 100mb) on the device and then pull it, measuring the time it takes. See below sample program. Default file size is 10 MB, second argument sets the test file size in MB.

```
/* Simple AFC test program (c) 2016 Nikias Bassen */
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <stdint.h>
#include <sys/time.h>

#include <libimobiledevice/libimobiledevice.h>
#include <libimobiledevice/afc.h>

static int timeval_subtract(struct timeval *result, struct timeval *x, struct timeval *y)
{
	// preserve *y
	struct timeval yy = *y;
	y = &yy;

	/* Perform the carry for the later subtraction by updating y. */
	if (x->tv_usec < y->tv_usec) {
		int nsec = (y->tv_usec - x->tv_usec) / 1000000 + 1;
		y->tv_usec -= 1000000 * nsec;
		y->tv_sec += nsec;	
	}
	if (x->tv_usec - y->tv_usec > 1000000) {
		int nsec = (y->tv_usec - x->tv_usec) / 1000000;
		y->tv_usec += 1000000 * nsec;
		y->tv_sec -= nsec;	
	}

	/* Compute the time remaining to wait.
	   tv_usec is certainly positive. */	
	result->tv_sec = x->tv_sec - y->tv_sec;
	result->tv_usec = x->tv_usec - y->tv_usec;

	/* Return 1 if result is negative. */
	return x->tv_sec < y->tv_sec;
}

int main(int argc, char** argv)
{
	int test_size = 10;
	int test_buf_size = 1048576;
	char *test_buf = NULL;

	idevice_t dev = NULL;
	afc_client_t afc = NULL;

	if (argc >= 2) {
		test_size = strtol(argv[1], NULL, 10);
	}
	idevice_new(&dev, NULL);

	if (!dev) {
		printf("Can't connect to device\n");
		return -1;
	}

	afc_client_start_service(dev, &afc, NULL);

	if (!afc) {
		idevice_free(dev);
		printf("Can't connect to AFC\n");
		return -1;
	}

	printf("Using test file size: %d MB\n", test_size);

	/* fill the buffer with random data */
	test_buf = malloc(test_buf_size);
	srand(time(NULL));
	int k;
	for (k = 0; k < test_buf_size; k++) {
		test_buf[k] = (k + rand()) % 256;
	}

	struct timeval tstart;
	struct timeval tend;
	struct timeval tdiff;

	/* write file to device */
	printf("### WRITE test with %d MB test file ###\n", test_size);
	gettimeofday(&tstart, NULL);
	uint64_t f = 0;
	afc_file_open(afc, "/test.tmp", AFC_FOPEN_RW, &f);
	uint64_t written = 0;
	uint64_t total = test_size*1048576;
	while (written < total) {
		uint32_t nwr = 0;
		afc_file_write(afc, f, test_buf, test_buf_size, &nwr);
		written+=nwr;
	}
	afc_file_close(afc, f);

	gettimeofday(&tend, NULL);
	timeval_subtract(&tdiff, &tend, &tstart);
	double duration = (double)tdiff.tv_sec + (double)tdiff.tv_usec/1000000.0f;
	printf(" -> Duration: %0.3f seconds, approx. WRITE speed: %0.2f MB/s\n", duration, (double)test_size/duration);

	/* get file size from device */
	char **info = NULL;
	uint64_t file_size = 0;
	if (afc_get_file_info(afc, "/test.tmp", &info) == AFC_E_SUCCESS) {
		int i;
		for (i = 0; info[i]; i += 2) {
			if (!strcmp(info[i], "st_size")) {
				file_size = atoll(info[i+1]);
				break;
			}
		}
	}
	afc_dictionary_free(info);

	if (file_size > 0) {
		/* read the file we wrote above */
		printf("### READ test with %d MB test file ###\n", (int)(file_size/1048576));
		gettimeofday(&tstart, NULL);
		f = 0;
		afc_file_open(afc, "/test.tmp", AFC_FOPEN_RW, &f);
		uint64_t bytes_read = 0;
		while (bytes_read < file_size) {
			uint32_t nrr = 0;
			afc_file_read(afc, f, test_buf, test_buf_size, &nrr);
			bytes_read+=nrr;
		}
		afc_file_close(afc, f);

		gettimeofday(&tend, NULL);
		timeval_subtract(&tdiff, &tend, &tstart);
		double duration = (double)tdiff.tv_sec + (double)tdiff.tv_usec/1000000.0f;
		printf(" -> Duration: %0.3f seconds, approx. READ speed: %0.2f MB/s\n", duration, (double)test_size/duration);
	}	

	/* remove test file from device */
	afc_remove_path(afc, "/test.tmp");
	afc_client_free(afc);

	idevice_free(dev);

	free(test_buf);

	return 0;
}
```
Sample output from a run on my Core i7 Macbook Pro:
```
$ ./afctest 100
Using test file size: 100 MB
### WRITE test with 100 MB test file ###
 -> Duration: 3.455 seconds, approx. WRITE speed: 28.95 MB/s
### READ test with 100 MB test file ###
 -> Duration: 4.097 seconds, approx. READ speed: 24.41 MB/s
```

mexmer, 2016-10-20:
which is why i suspect it has something to do with arm kernel optimization. or even might be hardware issue. there is lot of chinesse ARM cpus, that although have integrated USB 2.0 controller, cannot achieve even half of nominal USB 2 speed (which is 480Mbit/s)

zgoda-mobica, 2016-10-21:
Thanks @nikias here's what I get on ARMv7:

```
$ ./afctest 100
Using test file size: 100 MB
### WRITE test with 100 MB test file ###
 -> Duration: 6.256 seconds, approx. WRITE speed: 15.99 MB/s
### READ test with 100 MB test file ###
 -> Duration: 3.778 seconds, approx. READ speed: 26.47 MB/s
```

So while writing to iphone is significantly slower, read speed matches x86, I guess I can rule out USB2.0 bandwidth issue.

Unfortunately what's left is lack of processing power and that issue is out of my reach. I accept that [A20 is not a monster](https://linux-sunxi.org/A20). Thank you gents for making me understand where my problem is, I really appreciate your help.

I think it is safe to close this issue.

gitsop01, 2016-12-14:
@zgoda-mobica 
As you opened the issue you can also close it.

closed 2017-01-03 by zgoda-mobica