libimobiledevice/libimobiledevice #405: Version release (cash bounty)

txaj, 2016-10-19:
Hello, 

I am willing to wire 20 euros by SEPA to a European individual or to pledge to give 40 euros to a tax deductible european charity of the choosing of the contributor who can make a new release of this library before october 31th. 

The individual should leave its email address in the commit message so I can contact him.

If anyone wants to pledge some bucks with me, feel free to pitch in. 

This ticket duplicates at least #380 #340 and #401

Thanks.

KaiKemmann, 2016-10-22:
Good idea!
I did just donate 20€ to the [account](https://www.paypal.com/us/cgi-bin/webscr?cmd=_flow&SESSION=Cr30TRiB_iJ1SRXuvm8FeK37_fqzxM1-0kO03ABhaFp_p4qvsWKlHH6Zsqu&dispatch=5885d80a13c0db1f8e263663d3faee8d4fe1dd75ca3bd4f11d72275b28239088) given on the [libimobiledevice](http://www.libimobiledevice.org/) website the developer Martin Szuleki

fxcoudert, 2019-08-31:
@nikias 

Will there be a new release of libimobiledevice, including OpenSSL 1.1 support, before OpenSSL 1.0 reaches EOL on 2019-12-31? Otherwise, we will have to remove it from distributions like Homebrew.

nikias, 2019-08-31:
@fxcoudert, I am working on a release but currently there's an OpenSSL issue That really needs to be solved first (#817, #833). I am pushing hard to get it released within the upcoming weeks followed by a much shorter release cycle. Thanks.

FunkyM, 2020-06-15:
Closing as version [1.3.0](/libimobiledevice/libimobiledevice/releases/tag/1.3.0) has been released.

closed 2020-06-15 by FunkyM