libimobiledevice/libimobiledevice #406: Plist

Mradr, 2016-10-20:
I'm a python programmer and I seem to be having some issues with the 10.0.2 plist files. Seems some of them have been done in some sort of new format. Looking for a way to edit these plist files.

As for what's going on, it seems plistlib.readPlist("iPhoneBackups/%s/%s" % (phoneuuid, location))
returns that the file is not a plist file. Thought this was weird, so I went to python 3.5 to roll out binary as well, and same problem that the plist is not a plist. So I went to known plist files and same issue. Some of these files did load right - but a ton of them did not.

Not sure if it's something to do with how it might be backing up or maybe encrypting them? Any insite would be most helpful here.

nikias, 2016-10-20:
If there's a new format, please provide an example. Since you are talking about a backup, please mind that the backup might be encrypted.

Mradr, 2016-10-22:
After bashing my head into the wall I found that the file was encrypted. Problem fix for now :)

closed 2016-10-22 by Mradr