libimobiledevice/libimobiledevice #408: 'Waiting for device' idevicerestore.exe

ivanber7, 2016-10-22:
Hello to all,

I'm trying to restore a firmware CFW 9.3.2 in an iphone 6 and libimobiledevice show me the error "waiting for mobile" when the device is in recovery mode.

When the device is in mode DFU the error which show me is "sending ibec" is not an error, it stuck there.

I don't know what to do.

Anyone have had this error?

nikias, 2016-10-22:
Come on please. A custom firmware doesn't work on devices without a boot level exploit. Basically every device generation after iPhone 4 does not work with custom firmwares.
That's why it doesn't work.

closed 2016-10-22 by nikias

ivanber7, 2016-10-22:
Ahm ok, and how can i proceed?
Can you help me, please?

nikias, 2016-10-22:
You don't seem to understand. CFW on iPhone 6 doesn't work. Period. You can only restore official firmwares that are still signed by Apple.

ivanber7, 2016-10-22:
I'm spanish sorry for my english.
 
according to you, a custom firmware doesn't work on devices without a boot level exploit, by that i question you about that.

I have tried also with an original firmware but always is stuck in "waiting for device". 
it's rare. I don't know what to do, help is always welcome.

thank you for replies.
[image]

nikias, 2016-10-23:
As you can see here: https://ipsw.me/iPhone7,2 the only firmware that is currently being signed by Apple is iOS 10.0.2.
Also the hosts entry for gs.apple.com is not pointing to an Apple IP address. You should remove that entry so you can restore official firmwares.

ivanber7, 2016-10-24:
I try with & without hosts entry, with iOs 10.0.2, 9.3.2, 9.3.3, 9.3.5, all way but always show me error.

what i have to do bypass icloud? :/ because using cfw network will never work so..