libimobiledevice/libimobiledevice #411: Anyway to install new IPCC file by libimobiledevice ?

quangnhut123, 2016-10-26:
Now i need to test something with new ipcc file for my current network. But i wonder how to install it by my C# tool via this lib ?
Thanks

nikias, 2016-10-26:
ideviceinstaller can install ipcc.

quangnhut123, 2016-10-26:
But ideviceinstaller can work without iTunes and work with IOS 10 ?

nikias, 2016-10-26:
of course. it uses libimobiledevice: https://github.com/libimobiledevice/ideviceinstaller

nikias, 2016-10-26:
Just make sure you're using the latest code from git.

quangnhut123, 2016-10-26:
I see ideviceinstaller need some dependencies like : libimobiledevice, libplist, libzip
So how to i can compile with latest code from git in Visual Studio 2015 to make an exe and dll work in windows ?

nikias, 2016-10-26:
Of course it has dependencies! How would it work without libimobiledevice if it's using libimobiledevice? Visual Studio building is not supported. I mean, you can try it but it will probably fail. Building with MinGW works just fine.

quangnhut123, 2016-10-27:
MinGW i hear that name before but it can build whole this project by checkout and build to exe or dll can run on windows ?

nikias, 2017-06-18:
You can build all required libraries and tools with MinGW.

closed 2017-06-18 by nikias